KJ7LNW/xnec2c #1: File system syncs can wipe out performance

bbarenblat, 2021-12-09:
xnec2c makes a blocking call to [`sync`](https://pubs.opengroup.org/onlinepubs/9699919799/functions/sync.html) every time it closes a `FILE`. On certain file systems (notably ZFS), `sync` can take several seconds to execute, during which time xnec2c’s UI is completely unresponsive.

Git suggests the `sync` call got added in 3.9-beta. Do you know why it got put in originally? If the goal was crash-safety, there are more efficient ways to achieve that using `fsync`, `fdatasync`, etc.

bbarenblat, 2021-12-09:
Here’s the problematic call: https://github.com/KJ7LNW/xnec2c/blob/247d1b4a02f3c05d8abe05de8ec1a8261d43f4aa/src/utils.c#L307

KJ7LNW, 2021-12-09:
It was probably data safety, would you like to turn it into an `fsync(fd)` and send a PR?

KJ7LNW, 2021-12-11:
Thas has been updated this to fsync(fd) and a new version has been pushed.

Thanks for the report!

closed 2021-12-11 by KJ7LNW

bbarenblat, 2021-12-16:
You’re welcome ­– thank you for taking care of it!