KJ7LNW/xnec2c #3: Cannot compile from sources

scaramacai, 2021-12-18:
I tried with both the source from master and with the tar.gz of the 4.3.4 version and I got the following error:
Preparing the xnec2c build system...please wait

Found GNU Autoconf version 2.69
Found GNU Automake version 1.16.1
Found GNU Libtool version 2.4.6

Automatically preparing build ... Warning: autoreconf failed
Attempting to run the preparation steps individually

Preparing build ... configure.ac:79: error: required file './config.rpath' not found
configure.ac:79: error: required file './ABOUT-NLS' not found
ERROR: automake failed

I'm on Ubuntu 20.04

Thank you

Gigi

ewheelerinc, 2021-12-19:
I'm having trouble reproducing the exact error on my Ubuntu 20.04 system, but I have lots of extra packages installed.

Try this:

```sh
apt-get update
apt-get install gettext gettext-devel autopoint 
```
If that doesn't work, try `apt-get install gnulib`

If it still doesn't work, then send me the output of `dpkg -l` so I can troubleshoot.

I'll be out of town until after Christmas so I can continue to troubleshoot with you when I return.

ewheelerinc, 2021-12-19:
Actually it looks like gettext-devel isn't a thing, exclude that.

ewheelerinc, 2021-12-19:
I updated the documentation, added some checks in autogen.sh and pushed version 4.3.5.  See the top of this page for detail: https://www.xnec2c.org/

Hopefully that will help others if they have a similar problem.  If this worked for you then please close the issue.

scaramacai, 2021-12-19:
It looked like I was missing autopoint. The new version helped to fix this and now xnec2c 4.3.5 is compiling and running.

closed 2021-12-19 by scaramacai