KJ7LNW/xnec2c #4: wishlist: improve gnuplot save for VSWR and S11

scaramacai, 2021-12-19:
Hi,
first of all thank you for this great job. Having xnec2c improved and available on Github is a very good thing.
I have a suggestion that I marked as wishlist since it is probably not at high priority, but I'd like to have implemented.
I tried the S11 output by changing Frequency _Plots Show S11 checkbutton state from 0 to 1_.
While the S11 is plotted, the values are not saved on the related gnuplot file.  Furthermore, in the same file, the reference impedance is missing, too.
I know that starting from the imput impedance and the reference one, all parameters can be evaluated, but, in my opinion, to help visualizing from different points of view, and to speed up using gnuplot, all the parameters related to the input impedance, that is real and imaginary part of Zin, VSWR, S11, and the reference impedance, should always be saved. Anyway, it would be nice to have at least the magnitude of S11 and the reference impedance saved.
Or maybe a box to select the parameters to output cold be implemented?
Thank you again.
Gigi.

KJ7LNW, 2021-12-19:
I'll see what I can do when I return after Christmas.  I've not used the GNU plot before so maybe you can provide some more detail.

Questions:
- Is the gnuplot output just text, or graphical?
- Can you attach an example of what you are looking for?
- Can you define the difference between reference impedance and input impedance?  I know xnec2c lets you change Z0, do you just want the Z0 value displayed on the plot somewhere?
- How can I test on my side to generate the plots you are using?
- What do you mean by "speed up using gnuplot" ?

Any other expansion on the gnuplot topic would be helpful.

KJ7LNW, 2021-12-19:
Please also try File->Optimizer Output.  It generates a file `yourantenna.nec.csv` in the same directory as the original `.nec` file.  The CSV contains all of the fields that I could add to the gnuplot output, and you might find it useful as well.  Other fields could be added as well.

Note that while Optimizer Output is flagged that any _external_ change to the `.nec` (like from an editor) will auto-reload in xnec2c and re-write the resulting `.csv` file.

scaramacai, 2021-12-20:
Thank you very much for your comments.

Questions:

    Is the gnuplot output just text, or graphical?

It is just text.

Can you attach an example of what you are looking for?

I was just suggesting to collect the various values available into just one file. Since gnuplot can deal with multicolumn data, I think  is better to write a file with frequency, Zin (real, imag, magnitude, phase), VSWR  and S11.
I modified gnuplot.c to achieve this. The attached zip file contains the modified gnuplot.c together with an example file, containing the impedance values of the 2m_yagi.nec example

    Can you define the difference between reference impedance and input impedance? I know xnec2c lets you change Z0, do you just want the Z0 value displayed on the plot somewhere?
reference impedance = Z0
input impedance = input impedance of the antenna.
I just wanted the Z0 value written somewhere in the gnuplot file. Pls. see the example.

    How can I test on my side to generate the plots you are using?

    What do you mean by "speed up using gnuplot" ?
I just meant that since you already have all the data, you can plot one or more of them, without the need to recompute, for example |S11! from VSWR and so on.

[attach.zip](https://github.com/KJ7LNW/xnec2c/files/7747350/attach.zip)

Any other expansion on the gnuplot topic would be helpful.

KJ7LNW, 2021-12-28:
Can you do a pull request with your change so I can see the diff?

KJ7LNW, 2021-12-28:
Once I create the plot file with your change, how does that affect the users' use of the plot output?

Are there different gnuplot commands that are needed to select the graphs you wish to see?

scaramacai, 2021-12-29:
Created a pull request

KJ7LNW, 2021-12-29:
Pull https://github.com/KJ7LNW/xnec2c/pull/5

KJ7LNW, 2021-12-29:
Is this markup recognized by `gnuplot` somehow, or just a textual notation that you chose?

```
Real{Zin}[Ohm]
```

scaramacai, 2021-12-31:
This i just a textual notation.

ewheelerinc, 2022-01-17:
@scaramacai,

In terms of implementation, I would like to have a series of .gnuplot "programs" that take a data file as an argument.  Xnec2c will include these programs and provide a menu item to launch the program with the data file.  

For example:
```
gnuplot -e "filename='foo.data'" load_impedances.plg

plot filename using 1:2 with lines title "z_real", filename using 1:3 with lines title "z_imag"
```
[ https://stackoverflow.com/a/12330483/14055985 ]

It would be nice to use column headings so we don't have to worry about column re-ordering.  For example: http://www.gnuplot.info/demo_5.3/columnhead.html

Would you like to write and contribute the gnuplot "programs" that present this?

If so, what plot programs would you find useful?

-Eric

scaramacai, 2022-01-19:
I'm including a zip file with two scripts that could be useful for plotting 2D patterns. They can be improved a lot, but anyway seems to work. Note that I used data taken from nec2c for these experiments.
[2D_pattern_script_examples.zip](https://github.com/KJ7LNW/xnec2c/files/7895158/2D_pattern_script_examples.zip)

ewheelerinc, 2022-01-24:
Thanks for sending those over! I'll see what I can do...

KJ7LNW, 2022-03-08:
@scaramacai, all gnuplot columns are now exported during frequency plot exports in version 4.4.7.  Have a look at the changelog where your name appears: https://github.com/KJ7LNW/xnec2c/commits/master

There is more that could be done with this, like packaging gnuplot scripts for users, but I've not gotten to that yet.  If you have any other scripts that you like to use with these plots then send them over too and I'll see if I can include them in the future.

-Eric

scaramacai, 2022-03-09:
Dear Eric,
thank you very much for having considered my suggestions and having 
cited me as a contributor.

Possibly I have other useful scripts that could be included. Moreover, 
what do you think about python and matplotlib? Scripts could be launched 
without the need to directly use python and matplotlib, however those 
must be installed.

I would like to face two more aspects that could be interesting:
1) as you already mention some time ago, fork directly a gnuplot 
instance, instead of using it as an external program.

2) export data in vtk format so that they can be visualized in paraview. 
As for the 3D radiation pattern, this is not difficult and I have 
already implemented it in an old working copy. The same could be done 
for near field. However, including it into xnec2c means that some files 
must be modified, including the glade ui. I don't know if I have time, 
but I could try to follow a step by step approach. To visualize the 
antenna structure, instead, to my (very little) knowledge, the best way 
is to construct it directly in vtk: this would imply a dependence on vtk 
(and its requirements) for xnec2c, or scripting in python.

What do you think?

Kind regards.

Gian Luigi

KJ7LNW, 2022-03-09:
> thank you very much for having considered my suggestions and having cited me as a contributor. Possibly I have other useful scripts that could be included. 

You are welcome, I'm glad to have the help and feedback!

> Moreover, what do you think about python and matplotlib? Scripts could be launched without the need to directly use python and matplotlib, however those must be installed. 

That is a possibility.  How would you import the data from xnec2c?  Have a look at the functions [Save_FreqPlots_Touchstone](https://github.com/KJ7LNW/xnec2c/blob/master/src/gnuplot.c#L73) and [Save_RadPattern_Gnuplot_Data](https://github.com/KJ7LNW/xnec2c/blob/master/src/gnuplot.c#L165). Notice how the functions `meas_write_data` and `meas_write_format` are used, between the two you can create about any tabular output format that you could want.

> 2) export data in vtk format so that they can be visualized in paraview. As for the 3D radiation pattern, this is not difficult and I have already implemented it in an old working copy. The same [...]

Glade is pretty simple, just find a section that is similar and duplicate it to add a menu item.  The `<signal ... handler="c_function">` part specifies the callback and most callbacks are implemented in `src/callbacks.c`.

> To visualize the antenna structure, instead, to my (very little) knowledge, the best way is to construct it directly in vtk: this would imply a dependence on vtk (and its requirements) for xnec2c, or scripting in python. What do you think? Kind regards. Gian Luigi

Can you provide an example of how the antenna stucture would be represented in a .vtk file?  It is ok to make vtk or python an optional dependency, but only if we fork() the programs directly or dynamically link to them so xnec2c can run with minimal compile-time dependencies for those who do not have vtk or python dev libraries installed.

I think most of what you are looking to accomplish can be done by piping into existing programs or launching them against a data file.  Any binding code (like .gplot scripts or .py scripts) could be included in xnec2c in /usr/share/xnec2c/scripts/ so users can access them as needed.

Send me a pull request when you build something out to try!

KJ7LNW, 2023-07-20:
Inactive, closing issue.  Please re-open the issue or post a pull-request if you add start working on the features discussed above like Python and VTK.

closed 2023-07-20 by KJ7LNW