KJ7LNW/xnec2c #6: Build failing on Linux Mint/Ubuntu 18.04

m-anish, 2022-01-03:
```
checking for PACKAGE... no
configure: error: Package requirements (gtk+-3.0 >= 3.18.0) were not met:

No package 'gtk+-3.0' found

```

dpkg -l output : https://termbin.com/hlrq

Build failing at `./configure` step.

m-anish, 2022-01-03:
Relevant line from dpkg output

`ii  libgtk-3-0:amd64                                            3.22.30-1ubuntu4                                            amd64        GTK+ graphical user interface library`

m-anish, 2022-01-03:
Fixed: in ubuntu 18.04 `libgtk-3-dev` needs to be installed too. Perhaps a note in documentation.

KJ7LNW, 2022-01-03:
Thanks of the note, I'll add libgtk-3-dev to the package list for Ubuntu/Debian!

As for xnec2c.conf, please check a few things:
- Do you already have a /home/anish/.xnec2c directory?  
  - If so, is it writable by your user?
- Do you have an existing /home/anish/.xnec2c/xnec2c.conf file?
  - If so, is it writable by your user?
  - If you rename the file, does it generate a new one?
- Try changing changing Open_File in src/utils.c to report the actual error:
  - On this line: https://github.com/KJ7LNW/xnec2c/blob/master/src/utils.c#L286
  - Change it as follows and try again so it reports the error provided by the OS:
    ```c
    snprintf( mesg, sizeof(mesg), _("xnec2c[%d]: %s: Failed to open file: %s\n"), getpid(), fname, strerror(errno) );
    ```

KJ7LNW, 2022-01-04:
I just pushed v4.4.2 with the change above, it might provide a better error and indicate the actutal problem.  There are other stability updates in 4.4.2, so please update!

KJ7LNW, 2022-01-04:
Also, fyi: https://github.com/KJ7LNW/xnec2c/commit/0f35bc08e8757e3ffc606765974451f81508debb

m-anish, 2022-01-04:
> Thanks of the note, I'll add libgtk-3-dev to the package list for Ubuntu/Debian!
>
> As for xnec2c.conf, please check a few things:
>
>     * Do you already have a /home/anish/.xnec2c [...]

Thanks. I'll update the version. I don't remember exactly what I did, but now it is running!

KJ7LNW, 2022-01-04:
Glad its working!

 If you do antenna design you might check out https://github.com/KJ7LNW/xnec2c-optimize .

closed 2022-01-04 by KJ7LNW