KJ7LNW/xnec2c #8: isssues with single frequency FR

scaramacai, 2022-04-02:
I have issues to visualize currents and frequency parameters when a single frequency is indicated in FR card.
nec2c instead works perfectly.
I'm adding an example but this is a general behavior.
file1.nec does not work
file2.nec works
[example_files.zip](https://github.com/KJ7LNW/xnec2c/files/8402145/example_files.zip)

KJ7LNW, 2022-04-05:
Thanks for the bug report, fixed in 7977999.

By the way, do you have a call sign that you would like include with your name in the change comments?

scaramacai, 2022-04-07:
Thank you for fixing this issue.
I have no longer been a ham for several years, if that's what you mean.
Ciao.
Gian Luigi

KJ7LNW, 2022-04-07:
Glad it worked, enjoy your day!

closed 2022-04-07 by KJ7LNW