KJ7LNW/xnec2c #9: necsy

scaramacai, 2022-04-02:
I'm writing a python script (__necsy__) to preprocess nec files containing the SY directive.
The script is still disorganized and probably with bugs, but already works in many cases I tested.
If you are interested, find it at https://github.com/scaramacai/necsy

KJ7LNW, 2022-04-05:
Interesting.  Could this be made interactive so 'SY' lines could be passed by stdin and responses via stdout?

The xnec2c parser could add SY support by: 
  - forking a python script
  - write SY command to the python child
  - once a single SY is recieved all further processing would be sent to python before xnec2c reads the response and processes it.

Thus the python necsy code would be a co-processor for xnec2c when loading files.

Questions:
 - What do you think?
 - What SY features are missing from this implementation?
 - Are your .nec samples compatible with 4nec2? We should strive for parity with that existing implementation.

KJ7LNW, 2022-04-05:
Indeed, you could help remove this comment from input.c!

```c
case SY: /* "sy" TODO Compatibility with 4nec2.
            Too difficult, may never happen :-( */
  continue;

```

KJ7LNW, 2022-04-07:
> Your comment [#9 (comment)](https://github.com/KJ7LNW/xnec2c/issues/9#issuecomment-1089420841) is very interesting and ambitious. In my opinion the first thing we could try is to pipe the output of necsy to the input of xnec2c. It would require xnec2c being able to read from stdin; is this already possible?

It is if you disable forking because child processes expect to read the same input file.  Here is a quick hack:
```sh
cat examples/airplane.nec | xnec2c -i /proc/self/fd/0 -j0
```

The `-j0` makes it possible by disabling forking.

> At the moment i'm still investigating the capabilities of necsy to do the translation, without aborting or showing errors. I'm not a 4nec2 user and I have not yet tested any simulation results against 4nec2. I'll try to install it on a virtual machine and start testing.

FYI: 4nec2 works in WINE, too.

> As for the missing features:
> 
> * the FIX function is still to be implemented (but it should be trivial)
> 
> * AWG wire radii in mm are implemented for the standard default values in 4nec2 (awg20 to awg0). However the use of the # character in variables is not allowed in necsy and 4nec2's #0 ... #20 are replaced with AWG_0 ... AWG_20.
> 
> * 4nec2 allows for this syntax for measurement units and maybe in other places: number space unit. This is not possible in necsy. For example you need to write 20.5mm and not 20.5 mm
> 
> * Apart from the SY card, 4nec2 allows for an extended syntax in other places. Notably the percent sign used in EX cards to locate the feed segment; necsy is able to recognize this, but, at the moment, an EX line of this kind is simply copied verbatim, without further processing. While processing the other fields of the cards should be possible with a limited effort, I think that the percent syntax should, anyway, be processed by xnec2, because it implies the knowledge of the wire IDs, the number of segments and so on.

That makes sense, it is probably somewhat simple.  Can you create an example .NEC file so I can try to add that?

> Examples files are taken from 4nec2 and from a couple of other sources. I plan to test (Step 1: check only for the capability of providing an output) all the files bundled with 4nec2.

There are some other pre-processing that necsy could do, like EX type 6 and LD types 6 and 7:
* 4nec2: https://antennatestlab.com/wp-content/uploads/2017/05/4Nec2-Manual.pdf
    * EX type 6: page 21
        Internally in 4nec2 this EX type 6 is converted to a voltage-source on
        a dummy segment and connected to the initial segment through a network
        NT-card. To see the results of the internal current- to voltage source
        conversion, use ‘Show -> Nec 2 input’ on the ‘Geometry (F3)’
        window after a (4)Nec2 calculation is done or select the ‘show Nec’
        menubar option when using the ‘geometry editor’. Mostly the default
        Nec voltage-source (EX 0) is used. When modeling with multiple sources
        where the current for each source requires a specific phase relationship,
        the use of current-sources (EX 6) is needed.
    * LD type 6: page 22
    * LD type 7: page 22

KJ7LNW, 2025-12-29:
@scaramacai Artificial intelligence has made this reasonably feasible. 

Initial SY support is now available: Can you give it a try?