KJ7LNW/xnec2c #10: xnec2c exit with segfault

scaramacai, 2022-04-04:
I'm attaching a zip (examples.zip) with  files giving or not a segmentation fault. [examples.zip](https://github.com/KJ7LNW/xnec2c/files/8409939/examples.zip)
Moxon_900.nec works
Moxon_seg_fault.nec gives a segmentation fault
Moxon_XQ_and_0.nec gives a segmentation fault
Moxon_XQ_and_1.nec works
Moxon_without_RP_and_XQ.nec gives a segmentation fault

scaramacai, 2022-04-04:
I modified input.c in the XQ part, by moving the check:

        if( itmp1 == 0 )
          break; /* No rad pat, stop reading commands */

after the fpat initialization, and now both Moxon_seg_fault.nec and Moxon_XQ_and_0.nec work, while Moxon_without_RP_and_XQ.nec  still seg faults. Therefore I guess that fpat must be initialized anyway.

I'm attaching the (gzipped) modified file.

[input.c.modified.gz](https://github.com/KJ7LNW/xnec2c/files/8410117/input.c.modified.gz)

KJ7LNW, 2022-04-05:
Thanks for the report, fixed in 31b903e .

closed 2022-04-05 by KJ7LNW