KJ7LNW/xnec2c #19: Ubuntu 22.4.04.1 LTS install fails

RonKQ43, 2022-11-30:
fails with
[image]
started install with:
[image]
CPU: Intel Core i7-10870H @ 2.2GHz x 16
OS Name Ubuntu 22.04.1 LTS
OS Type: 64 bit
Gnome Ver: 42.5
Windowing System: Wayland

KJ7LNW, 2022-11-30:
Hi Ron, I'm guessing its a permissions issue and I think it was fixed in a newer release (8e096c9c3ca47a35b9d131e119a0d5718a65dd99).  

Can you send me the output this:

```sh
ls -lR ~/.xnec2c
```

And then try this to fix it after sending me the `ls` output from above:

```
sudo chown ronq.ronq -R ~/.xnec2c
chmod 755 ~/.xnec2c
chmod 644 ~/.xnec2c/*
```

KJ7LNW, 2022-12-28:
Hi @RonKQ43, did the commands above solve the issue?

KJ7LNW, 2023-01-10:
Hi @RonKQ43,

I am closing this issue but feel free to reopen it if the problem persists.

-Eric, KJ7LNW

closed 2023-01-10 by KJ7LNW