KJ7LNW/xnec2c #23: OpenGL Support

KJ7LNW, 2022-12-01:
HELP WANTED
============

Would you like to volunteer to help with OpenGL support to xnec2c?  Read on:

This first-pass implementation partially renders antenna patterns in OpenGL.  Hurray! While the OpenGL work has been started, it needs some polish (ahem, quite a bit of polish)...plus, this is a feature that the original xnec2c author, [Neoklis 5B4AZ](http://www.5b4az.org/), has always wanted.

-Eric, KJ7LNW

GOALS!
=======
- OpenGL for:
   - Radiation Pattern
   - E/H Field Pattern and animation
   - Physical Structure
- Click to rotate the object
- Mouse wheel to zoom
- Middle click to pan
- Shift+click to rotate light sources around the object
- Transparency when clicking to rotate.
- An optional transparent grid plane at (x,y)=0
- Shaders
   - I'm open to any cool shader ideas you might have
   - It could be neat to have one or more light sources such that the light sources can rotate around the radiation pattern (or structure) with shift-click+drag.

Getting Started
===============

```sh
git clone https://github.com/KJ7LNW/xnec2c.git
cd xnec2c
git checkout opengl
./autogen.sh
./configure
make
./src/xnec2c -j4 examples/airplane.nec
```

Once it loads:

- Enable the "View->Radiation Pattern" checkbox and click "Gain Pattern" button on the "Radiation Patterns" window
- When it loads there should be a small "GLArea Test" window.  Resize the window to a reasonable size for your display.
- The "GLArea Test" window will only update when "Radiation Patterns" is rotated.
- To interact with the GLArea Test window, click and drag in the "Radiation Patterns" window (_not_ the "GLArea Test" window) to change the OpenGL view.
- To change the scale, use your mouse wheel to zoom over the "Radiation Patterns" window (_not_ the "GLArea Test" window) and then rotate it so the GL window will redraw.

Development Considerations
=====================

There is a work-in-progress branch for OpenGL support using [CGLM](https://github.com/recp/cglm) which you can see in this commit comparison: https://github.com/KJ7LNW/xnec2c/compare/master...opengl

1. A new opengl.c and .h file should be created to store code for:
  - camera
  - arcball
  - vertex generation from NEC2 structures
  - GTK UI hooks for pan/scale/rotate
  - other GL stuff
2. A function from opengl.c for generating radition pattern vertexes should be hooked into `src/draw_radiation.c:Draw_Radiation_Pattern()`.  Currently that logic is hard-coded into `Draw_Radiation_Pattern()` but that was just a convenient place to put it while testing.  See the commits to understand what changed.
3. The current triangle creation logic isn't quite right and the radiation pattern isn't so clean.  Note that theta is the azimuth and theta=0 is pointing in the direction of Z+!  This is just how NEC2 was designed back in 1981...
4. The `configure.ac` file should detect OpenGL libraries and link them as necessary.  Some of this has been done already.
5. CGLM should be a git submodule if it is not already
   - If you prefer a different vector/matrix math library vs CGLM then I'm open to it; it looked like CGLM was a good place to start since xnec2c is written in C, not in C++.
6. Maintain shader version 1.20 support for older distributions, right now we don't need anything fancy.
7. To the extent possible, widgets should be maintained as GTK Builder objects and expressed as XML in `resources/xnec2c.glade`.  However, it might be easier to open-code as a first pass with gtk API calls and refactor into XML later.  
8. Rebase on the master branch
9. Try to follow [Linux Kernel commit practices and coding style](https://www.kernel.org/doc/html/v4.10/process/coding-style.html) for new code.

References
=========
- http://www.opengl-tutorial.org/beginners-tutorials/tutorial-6-keyboard-and-mouse/
- https://asliceofrendering.com/camera/2019/11/30/ArcballCamera/
- https://learnopengl.com/Getting-started/Camera
- https://cglm.readthedocs.io/en/latest/

jerryd, 2023-11-03:
I was thinking about doing some work to convert to GTK4. Has anyone considered doing this? KF7RCC

KJ7LNW, 2023-11-03:
Generally speaking, we want to support deployment on any system that is still supported under long-term support by the distribution vendor. For example Oracle Linux 8 is supported until 2029, and it runs GTK 3. That means we cannot cut over to GTK 4 without backwards compatibility support.

If there are GTK4 features that would be nice to add, then that would be great, however they need to be detected by Autoconf in `configure.ac` and appropriate `#ifdef` checks need to be placed in the code so that it will still compile on older distributions.