KJ7LNW/xnec2c #29: Flatpak build

AsciiWolf, 2023-04-13:
I have made a [Flatpak](https://flatpak.org) build of Xnec2c that I am going to publish on [Flathub](https://flathub.org). Feel free to let me know whether you are interested in co-maintaining the Flathub Flatpak and having push access to its Git repository. :-)

KJ7LNW, 2023-04-13:
That is great, thank you for publishing! 

Questions:

- Is this something where the flathub XML file could be maintained in xnec2c's git tree, or are they necessarily separate repositories? (for example, we already have a [xnec2c.spec](https://github.com/KJ7LNW/xnec2c/blob/master/xnec2c.spec) file for building RPMs).
- How does auto-update work?  If we publish a new version of xnec2c [on the releases index](https://www.xnec2c.org/releases/?C=M;O=D), will flathub magically pick it up?

AsciiWolf, 2023-04-13:
Hi Eric,

Thanks for the quick reply!

ad 1. It is possible for third-parties to have their own infrastructure, however it is not a common thing and I am not sure how hard it would be to set it up. You can try asking [here](https://github.com/flathub/flathub/pull/4070).
ad 2. Well, either the Flatpak maintainer can update the Flatpak by manually editing its (yaml) manifest file or there is something called [x-data-checker](https://github.com/flathub/flatpak-external-data-checker) bot that basically sends a new Pull Request to the [Flathub](https://github.com/flathub) app repository with updated version number, sha256 checksum, AppStream metadata version and other things when it detects a new upstream release. However, it requires specific metadata in the application Flatpak manifest file and I did not add these metadata yet. I will do it later. :-)

KJ7LNW, 2023-04-14:
That sounds great.  Let me know when you have the metadata plugged in and I'll take a look

AsciiWolf, 2023-04-14:
I will let you know. :-) If you want, let us also know [here](https://github.com/flathub/flathub/pull/4070) whether you are interested in having push rights to the Xnec2c Flatpak Git repository (once it is created) so you can merge the Pull Requests and do other changes to the Flatpak.

KJ7LNW, 2023-04-14:
Ok.  I commented on the thread for push rights.  Thanks!

KJ7LNW, 2023-07-20:
@AsciiWolf, I think this issue is ready to close, but please re-open if I missed something.

Thanks!

-Eric

KJ7LNW, 2024-04-06:
@AsciiWolf, we just released v4.4.13.  Is there anything that needs to be done to update flathub?

AsciiWolf, 2024-04-06:
@KJ7LNW I will take a look tomorrow. Thanks for letting me know!

AsciiWolf, 2024-04-06:
No need to reopen this ticket. xnec2c has been on Flathub for a while now and I have created a [separate ticket](https://github.com/flathub/org.xnec2c.Xnec2c/issues/7) for the update.

KJ7LNW, 2024-04-06:
got it

closed 2024-04-06 by KJ7LNW