KJ7LNW/xnec2c #30: Received `OS openbsd 7.3 not supported`. Is there any reason for this?

anoduck, 2023-04-20:
Before I attempt to see if I can troubleshoot the build process, I wanted to inquire as to if there were any incompatibility issues or known restrictions preventing a successful build.

```bash
checking for a BSD-compatible install... /usr/bin/install -c
checking whether build environment is sane... yes
checking for a thread-safe mkdir -p... /usr/local/bin/gmkdir -p
checking for gawk... gawk
checking whether make sets $(MAKE)... yes
checking whether make supports nested variables... yes
checking whether to enable maintainer-specific portions of Makefiles... no
checking build system type... x86_64-unknown-openbsd7.3
checking host system type... x86_64-unknown-openbsd7.3
configure: error: "OS openbsd7.3 is not supported"
```

cheers.

KJ7LNW, 2023-04-24:
@anoduck thanks for writing.  I'm not aware of any specific problem, it should be fine if you have the libraries that it needs.

I just changed that error to a warning (9969638), so pull the latest git master branch and see if that helps.  

If you still have trouble, then you might try merging this branch from PR #17 and see if it is more successful in OpenBSD:
- https://github.com/ndim/xnec2c/tree/buildsystem-cleanups

@ndim put a bunch of work into the build process cleanup, so I would like to merge his work when it is ready.

anoduck, 2023-04-27:
@KJ7LNW Awesome sauce. Thanks for the info.

KJ7LNW, 2023-04-27:
You're welcome, thanks for testing on OpenBSD.  Note the `-j` option for parallelism, too.  If you have OpenBLAS, ATLAS, or Intel MKL then it would be nice to test those for accelerated field calculations, too.  See [Accelerated Linear Algebra Support](https://www.xnec2c.org/#Linalg) in the docs for more detail.

KJ7LNW, 2023-04-27:
I should note that accelerated libraries are detected and dynamically loaded at runtime, not linked at compile time.  Running `xnec2c -vvv` will show the detection process, and library names for auto-detection with `dlopen()` are listed [here](https://github.com/KJ7LNW/xnec2c/blob/master/src/mathlib.c#L28).

anoduck, 2023-04-30:
Sometimes running OpenBSD is like riding into battle in an impenetrable suit of armor with both hands tied behind your back. You know you're protected, but unfortunately, you also know you aren't going to get anything accomplished either.

Compilation was a rather short affair.

```bash
ld: error: undefined symbol: g_module_supported
>>> referenced by interface.c:53
>>>               interface.o:(Gtk_Builder)

ld: error: undefined symbol: backtrace
>>> referenced by utils.c:806
>>>               utils.o:(print_backtrace)
>>> referenced by utils.c:806
>>>               utils.o:(_get_backtrace)

ld: error: undefined symbol: backtrace_symbols
>>> referenced by utils.c:807
>>>               utils.o:(print_backtrace)
>>> referenced by utils.c:807
>>>               utils.o:(_get_backtrace)
cc: error: linker command failed with exit code 1 (use -v to see invocation)
*** Error 1 in target 'xnec2c', line 455 of Makefile (continuing)
Error target(s) unaccounted for: all-am xnec2c all 
`all' not remade because of errors.
Errors while building
```

I just built it on Kali Linux on my development box. There is a lot of learning needed in order to fully employ the features. I don't want to make anyone else's life difficult, by troubleshooting OpenBSD build errors right now. Not before I fully learn how to employ the software.

KJ7LNW, 2023-04-30:
I just pushed two commits that may help:

1. c946d5c Actually check if backtrace exists.
2. d8743c1 Add g_module_supported detection

Item 1 is fine, you don't need backtrace() except for debugging...but item 2 might be a showstopper.  It allows it to compile and run, but it may not work at all unless GTK is magic on the OpenBSD platform.

If it doesn't work, then I would be curious to know what console output you get when you run this:

```sh
xnec2c -vvvd
```

anoduck, 2023-05-05:
Holy crap! It is building! Go baby, go... Well, that was exciting there for a moment. 

Still learning, but able to provide debugging of the make process, since that came so close to completion. 

```bash
Putting child 0x8de4c8137e0 (xnec2c) PID 77010 on the chain.
  Recipe of 'xnec2c' is being run.
 The prerequisites of 'all-am' are being made.
The prerequisites of 'all' are being made.
Live child 0x8de4c8137e0 (xnec2c) PID 77010 
Reaping winning child 0x8de4c8137e0 PID 77010 
egcc --pedantic -Wall -std=gnu11 -O2 -g -Wformat -Werror=format-security -fpie -Wno-overlength-strings -DGTK_DISABLE_SINGLE_INCLUDES -DGDK_DISABLE_DEPRECATED -DGTK_DISABLE_DEPRECATED -DGSEAL_ENABLE -DGDK_PIXBUF_DISABLE_DEPRECATED -DG_DISABLE_DEPRECATED -D_FORTIFY_SOURCE=2 -fcx-fortran-rules -fstack-protector-strong -fno-signed-zeros -fno-math-errno -Wl,--export-dynamic -Wl,--as-needed  -o xnec2c main.o mathlib.o measurements.o interface.o callbacks.o console.o callback_func.o calculations.o cmnd_edit.o geom_edit.o gnuplot.o draw.o draw_structure.o draw_radiation.o fields.o fork.o geometry.o ground.o xnec2c.o input.o matrix.o utils.o nec2_model.o network.o optimize.o plot_freqdata.o radiation.o rc_config.o shared.o somnec.o xnec2c-resources.o -L/usr/local/lib -Wl,-rpath-link,/usr/X11R6/lib -lgtk-3 -lgdk-3 -lpangocairo-1.0 -lpango-1.0 -lharfbuzz -latk-1.0 -lcairo-gobject -lcairo -lgdk_pixbuf-2.0 -lgio-2.0 -lgobject-2.0 -lglib-2.0 -lintl /usr/local/lib/libintl.so.7.0 -L/usr/local/lib /usr/local/lib/libiconv.so.7.1 -Wl,-rpath,/usr/local/lib -lpthread -lgmodule-2.0 -lm 
Live child 0x8de4c8137e0 (xnec2c) PID 99608 
main.c:129(main.o:(main)): warning: strcpy() is almost always misused, please use strlcpy()
main.c:79(main.o:(main)): warning: stpcpy() is dangerous; do not use it
ld: error: undefined symbol: backtrace
>>> referenced by utils.c:806
>>>               utils.o:(_get_backtrace)
>>> referenced by utils.c:806
>>>               utils.o:(print_backtrace)

ld: error: undefined symbol: backtrace_symbols
>>> referenced by utils.c:807
>>>               utils.o:(_get_backtrace)
>>> referenced by utils.c:807
>>>               utils.o:(print_backtrace)
collect2: error: ld returned 1 exit status
Reaping losing child 0x8de4c8137e0 PID 99608 
gmake[2]: *** [Makefile:455: xnec2c] Error 1
Removing child 0x8de4c8137e0 PID 99608 from chain.
Considering target file 'all'.
 File 'all' does not exist.
 Considering target file 'all-am'.
  File 'all-am' does not exist.
  Considering target file 'Makefile'.
  File 'Makefile' was considered already.
  Considering target file 'xnec2c'.
  Recently tried and failed to update file 'xnec2c'.
 Finished prerequisites of target file 'all-am'.
 Giving up on target file 'all-am'.
Finished prerequisites of target file 'all'.
Giving up on target file 'all'.
gmake[2]: Target 'all' not remade because of errors.
gmake[2]: Leaving directory '/home/user/xnec2c/src'
```

The problematic line is 806-807 in `/utils/src`. It does not recognize `backtrace`.

```c++
size = backtrace(array, 10);
strings = backtrace_symbols(array, size);
```

KJ7LNW, 2023-05-08:
Hmm, `HAVE_BACKTRACE` is flagged as 1. 

After you did a `git pull`, did you redo `./autogen.sh && ./configure` ?  ... because the commit c946d5c224bc0171c7e1854dc4c85ace596f57c8 should fix that but requires a new autogen call.

If it still doesn't work then please send me your `config.log` and `config.h` so I can figure out what is happening with the false positive.

After trying the things above: 

You could hack `HAVE_BACKTRACE` out of config.h (after `./autogen.sh && ./configure`) and see if it works.  This is what it should look like on your system:
```c
]$ grep HAVE_BACKTRACE config.h
/* #undef HAVE_BACKTRACE */
```

anoduck, 2023-05-10:
@KJ7LNW Well, I must be a monkey's uncle, because after starting over with a fresh clone, She is looking good. The only thing hanging is the last command in the Makefile on line 784. 

```bash
--sourcedir=../resources --target=xnec2c-resources.c --generate-source ../resources/xnec2c.gresource.xml
/bin/sh: /bin/sh: --: unknown option
*** Error 1 in target 'xnec2c-resources.c', line 785 of Makefile (continuing)
```

KJ7LNW, 2023-05-10:
I think that was fixed in @ndim's build cleanup.  I just merged the openbsd changes with fixes from @ndim into a **new openbsd branch** so you can test.  Please try this, modify as you need, but make sure you are on the `origin/openbsd` branch:

```sh
git clone https://github.com/KJ7LNW/xnec2c.git
cd xnec2c/
git checkout origin/openbsd
./autogen.sh 
./configure 
make
./src/xnec2c examples/airplane.nec -j4 -vvvd
```

anoduck, 2023-05-13:
Super Dooper Fantasticness!!! What a beautiful compilation. The warning count was incredibly low.

KJ7LNW, 2023-05-13:
Awesome!  So it loads the airplane model?  

Try these and make sure they work:
- View->Frequency Plots
   - Enable Max Gain and VSWR
   - Click the triangle "play" button
- Main Window->View->Radiation Pattern
   - Click "Gain Pattern
   - Rotate the 3D gain pattern

If those work then I think its working!

Note that you can pass `-j 20` or however many CPUs you have for parallel frequency computation.

Let me know if you have any other issues.

anoduck, 2023-05-15:
Unfortunately, if you want to run OpenBLAS or LAPACKe in OpenBSD, you have to downgrade your installation of gcc-8 due to available LLVM options. If I downgrade, then I will lose the lua-language-server, which requires gcc-11. 

So, as long I disable pthreads, it works fine.

KJ7LNW, 2023-05-15:
Interesting.  

Would you please clarify what do you mean by "disable pthreads"?

Questions:

1. Does it work only if you run `xnec2c --no-pthreads`?
    - If so then the UI will be choppy while running the frequency loop.  
    - Also, then I'm curious why it is necessary because It looks like [OpenBSD supports pthreads.](https://man.openbsd.org/pthreads.3)
3. Did you somehow compile _without_ pthreads, as in `./configure --disable-pthreads`?  If so, I didn't know that was an option.
4. Did you have to select File->Mathlib->Interactive/Batch->"NEC2 Gaussian Elimination" (instead of "OpenBLAS+LAPACKe pthreads")?

anoduck, 2023-05-16:
Without the `-P` flag it crashes when play was clicked. 

1. Yes, only with `-P`
  A. Nothing notice-able yet. 
  B. You are correct, pthreads are supported. Why Xnec2c is not able to facilitate this, is well beyond my skill set. OpenBSD is like that… difficult to work with on the best of days.
2. No, sir. I compiled solely with `./configure`
4. No, sir. No extra configuration was needed. 

It does squawk at me a little. 
```bash
[err] ATLAS, Threaded was not detected: Unable to set the mathlib index to 0
[err] ATLAS, Threaded was not detected: Unable to set the mathlib index to 0
```

KJ7LNW, 2023-06-01:
I wonder why it crashes.  Can you try this:

```sh
CFLAGS='-O0 -g' ./configure
make
gdb -ex run ./src/xnec2c
```

Get it to crash as you indicated (because there is no `-P`). When it crashes, can you run these commands at the `(gdb)` prompt and send me the output?

```sh
Segfault
(gdb) info locals
(gdb) info args
(gdb) bt
(gdb) thr a a bt
```

anoduck, 2023-06-03:
Yes, Sir.

(elevator music)

Built, and inducing segmentation fault.

```c++
Thread 8 (thread 540703):
#0  futex () at /tmp/-:3
#1  0x2d75572a5768b943 in ?? ()
#2  0x00000ef57b540455 in _twait (p=0xef5c614e870, val=9, clockid=3, abs=0xef5d7912dd0) at /usr/src/lib/libc/thread/synch.h:36
#3  _rthread_cond_timedwait (cond=0xef5c614e870, mutexp=0xef5c614ec40, abs=0xef5d7912dd0) at /usr/src/lib/libc/thread/rthread_cond.c:106
#4  0x00000ef6303e4ab4 in g_cond_wait_until (cond=<optimized out>, mutex=0xef620a3e980, end_time=<optimized out>) at ../glib-2.76.3/glib/gthread-posix.c:930
#5  0x00000ef63033e5a6 in g_async_queue_pop_intern_unlocked (queue=0xef620a3e980, wait=1, end_time=274239830055) at ../glib-2.76.3/glib/gasyncqueue.c:428
#6  0x00000ef63033e799 in g_async_queue_timeout_pop (queue=0xef620a3e980, timeout=15000000) at ../glib-2.76.3/glib/gasyncqueue.c:551
#7  0x00000ef6303b2137 in g_thread_pool_wait_for_new_pool () at ../glib-2.76.3/glib/gthreadpool.c:181
#8  g_thread_pool_thread_proxy (data=<optimized out>) at ../glib-2.76.3/glib/gthreadpool.c:408
#9  0x00000ef6303b0ae9 in g_thread_proxy (data=0xef5a53bb0f0) at ../glib-2.76.3/glib/gthread.c:831
#10 0x00000ef555761195 in _rthread_start (v=<optimized out>) at /usr/src/lib/librthread/rthread.c:96
#11 0x00000ef57b558ada in __tfork_thread () at /usr/src/lib/libc/arch/amd64/sys/tfork_thread.S:86

Thread 7 (thread 474925):
#0  _thread_sys_poll () at /tmp/-:3
#1  0xf7406100d0d4a754 in ?? ()
#2  0x00000ef57b5829f2 in _libc_poll_cancel (fds=0xef57fa09410, nfds=1, timeout=-1) at /usr/src/lib/libc/sys/w_poll.c:27
#3  0x00000ef63037c702 in g_main_context_poll (context=0xef5939d9f00, timeout=-1, fds=0xef57fa09410, n_fds=1, priority=<optimized out>) at ../glib-2.76.3/glib/gmain.c:4584
#4  g_main_context_iterate (context=0xef5939d9f00, block=<optimized out>, dispatch=<optimized out>, self=<optimized out>) at ../glib-2.76.3/glib/gmain.c:4271
#5  0x00000ef63037c80b in g_main_context_iteration (context=0xef5939d9f00, may_block=1) at ../glib-2.76.3/glib/gmain.c:4343
#6  0x00000ef5dad9a7cd in dconf_gdbus_worker_thread (user_data=0xef5939d9f00) at ../dconf-0.40.0/gdbus/dconf-gdbus-thread.c:82
#7  0x00000ef6303b0ae9 in g_thread_proxy (data=0xef5c767c230) at ../glib-2.76.3/glib/gthread.c:831
#8  0x00000ef555761195 in _rthread_start (v=<optimized out>) at /usr/src/lib/librthread/rthread.c:96
#9  0x00000ef57b558ada in __tfork_thread () at /usr/src/lib/libc/arch/amd64/sys/tfork_thread.S:86

Thread 6 (thread 243123):
#0  _thread_sys_poll () at /tmp/-:3
#1  0xf7406100d0d4a754 in ?? ()
#2  0x00000ef57b5829f2 in _libc_poll_cancel (fds=0xef5ca7526a0, nfds=2, timeout=-1) at /usr/src/lib/libc/sys/w_poll.c:27
#3  0x00000ef63037c702 in g_main_context_poll (context=0xef5bed6e6c0, timeout=-1, fds=0xef5ca7526a0, n_fds=2, priority=<optimized out>) at ../glib-2.76.3/glib/gmain.c:4584
#4  g_main_context_iterate (context=0xef5bed6e6c0, block=<optimized out>, dispatch=<optimized out>, self=<optimized out>) at ../glib-2.76.3/glib/gmain.c:4271
#5  0x00000ef63037c80b in g_main_context_iteration (context=0xef5bed6e6c0, may_block=1) at ../glib-2.76.3/glib/gmain.c:4343
#6  0x00000ef63037e7f1 in glib_worker_main (data=<optimized out>) at ../glib-2.76.3/glib/gmain.c:6455
#7  0x00000ef6303b0ae9 in g_thread_proxy (data=0xef5bed6dd70) at ../glib-2.76.3/glib/gthread.c:831
#8  0x00000ef555761195 in _rthread_start (v=<optimized out>) at /usr/src/lib/librthread/rthread.c:96
#9  0x00000ef57b558ada in __tfork_thread () at /usr/src/lib/libc/arch/amd64/sys/tfork_thread.S:86

Thread 5 (thread 479798):
#0  futex () at /tmp/-:3
#1  0x2d75572a5768b943 in ?? ()
#2  0x00000ef57b540455 in _twait (p=0xef62159b640, val=27, clockid=3, abs=0x0) at /usr/src/lib/libc/thread/synch.h:36
#3  _rthread_cond_timedwait (cond=0xef62159b640, mutexp=0xef5c614e5b0, abs=0x0) at /usr/src/lib/libc/thread/rthread_cond.c:106
#4  0x00000ef6303e4846 in g_cond_wait (cond=<optimized out>, mutex=0xef5c6140e40) at ../glib-2.76.3/glib/gthread-posix.c:794
#5  0x00000ef63033e585 in g_async_queue_pop_intern_unlocked (queue=0xef5c6140e40, wait=1, end_time=-1) at ../glib-2.76.3/glib/gasyncqueue.c:425
#6  0x00000ef6303b11f8 in g_thread_pool_spawn_thread (data=<optimized out>) at ../glib-2.76.3/glib/gthreadpool.c:311
#7  0x00000ef6303b0ae9 in g_thread_proxy (data=0xef5c6151410) at ../glib-2.76.3/glib/gthread.c:831
#8  0x00000ef555761195 in _rthread_start (v=<optimized out>) at /usr/src/lib/librthread/rthread.c:96
#9  0x00000ef57b558ada in __tfork_thread () at /usr/src/lib/libc/arch/amd64/sys/tfork_thread.S:86

Thread 4 (thread 206009):
#0  thrkill () at /tmp/-:3
#1  0xadbaaf97d07b4a5e in ?? ()
#2  0x00000ef57b5495d2 in _libc_abort () at /usr/src/lib/libc/stdlib/abort.c:51
#3  0x00000ef57b5cc51f in _rthread_mutex_trylock (mutex=<optimized out>, trywait=0, abs=0x0) at /usr/src/lib/libc/thread/rthread_mutex.c:117
#4  _rthread_mutex_timedlock (mutexp=<optimized out>, trywait=0, abs=0x0, timed=<optimized out>) at /usr/src/lib/libc/thread/rthread_mutex.c:167
#5  0x00000ef6303e3d8a in g_mutex_lock (mutex=0xef5ca751260) at ../glib-2.76.3/glib/gthread-posix.c:224
#6  0x00000ef349a2a027 in _g_idle_add_once (function=0xef349a165f0 <update_freq_mhz_spin_button_value>, data=0xef5c768d670, lock=1) at utils.c:747
#7  0x00000ef349a2a093 in g_idle_add_once_sync (function=0xef349a165f0 <update_freq_mhz_spin_button_value>, data=0xef5c768d670) at utils.c:767
#8  0x00000ef349a17835 in Frequency_Loop_Thread (p=0x0) at xnec2c.c:984
[... cut ...]
#10 0x00000ef57b558ada in __tfork_thread () at /usr/src/lib/libc/arch/amd64/sys/tfork_thread.S:86

Thread 3 (thread 324715):
Couldn't get registers: No such process.
```

-----

It might be worth noting I am receiving the following error during launch. 

```bash
(xnec2c:52104): GVFS-WARNING **: 03:35:23.764: The peer-to-peer connection failed: Error when getting information for file “/tmp/runtime-user/gvfsd”: No such file or directory. Falling back to the session bus. Your application is probably missing --filesystem=xdg-run/gvfsd privileges.
```

KJ7LNW, 2023-06-05:
It looks like locking a lock that was already held in the same thread will crash on *BSD.  Hmm... 

I use double-locking as a way to wait for another thread to finish, so when it unlocks, then I get the lock and continue.   Apparently that (maybe hackish) synchronization method is not portable.  I think `g_cond_wait` is the better way to do this, but it's going to take some troubleshooting to get right...and I should check the code for any other lock/lock/unlock cases, too.

For now use `-P` to work around the problem, but this really is a bug.

Cool find, now to see what the solution might be.  Notes:

```
#0  thrkill () at /tmp/-:3
#1  0x78c61e1787f0e1a6 in ?? ()
#2  0x00000309b2217dee in _libc_abort () at /usr/src/lib/libc/stdlib/abort.c:51
#3  0x00000309b21f637f in _rthread_mutex_timedlock (mutexp=Variable "mutexp" is not available.
) at /usr/src/lib/libc/thread/rthread_mutex.c:117
#4  0x00000309ad5e5c96 in g_mutex_lock () from /usr/local/lib/libglib-2.0.so.4201.9
#5  0x000003076903f5ed in _g_idle_add_once (function=0x3076902bcb0 <update_freq_mhz_spin_button_value>, 
    data=0x30a624ef910, lock=1) at utils.c:749
#6  0x000003076903f65f in g_idle_add_once_sync (function=0x3076902bcb0 <update_freq_mhz_spin_button_value>, 
    data=0x30a624ef910) at utils.c:769
#7  0x000003076902cef1 in Frequency_Loop_Thread (p=0x0) at xnec2c.c:984
#8  0x0000030a0c8a5d81 in _rthread_start (v=Unhandled dwarf expression opcode 0xa3
) at /usr/src/lib/librthread/rthread.c:96
#9  0x00000309b21fbd6a in __tfork_thread () at /usr/src/lib/libc/arch/amd64/sys/tfork_thread.S:86
#10 0x00000309b21fbd6a in __tfork_thread () at /usr/src/lib/libc/arch/amd64/sys/tfork_thread.S:86
```

The fix: I need a condition variable between utils.c:747 and utils.c:699 (or try a recursive mutex).

Documentation:
- https://docs.gtk.org/glib/struct.RecMutex.html
- https://www.manpagez.com/html/glib/glib-2.42.1/glib-Threads.php#GCond
- https://docs.gtk.org/glib/method.Cond.wait.html
- https://pubs.opengroup.org/onlinepubs/7908799/xsh/pthread_mutexattr_settype.html

KJ7LNW, 2023-06-07:
I think commit 04d15ac861ceb1b27b165e206b66c799df86f42c fixes the crash.  Please try the latest git version _without_ the `-P` option.

KJ7LNW, 2023-06-09:
The `openbsd` branch is old now (deleted), I released this fix to `master`.

Here is how I test locally:

```sh
git pull origin # or git checkout origin/master
cat INSTALL.OPENBSD
# pkg_add autoconf-2.71 automake-1.16.5 gettext-tools-0.21.1 gtk3mm-3.24.7
# export AUTOMAKE_VERSION=1.16 AUTOCONF_VERSION=2.71 MAKE=gmake
# ./autogen.sh
# ./configure
# gmake
# ./src/xnec2c examples/airplane.nec
```

anoduck, 2023-06-09:
That explains it, and yes, that works very nicely indeed. Excellent work. If you need anything at all, let me know.

closed 2023-06-09 by anoduck

KJ7LNW, 2023-06-09:
Thanks for your help!  Now xnec2c can claim that it officially supports OpenBSD.

By the way, can you share how you will be using xnec2c?  
Are you a ham, and if so, what is your call sign?

anoduck, 2023-06-09:
> Thanks for your help! Now xnec2c can claim that it officially supports OpenBSD.

Now I need to work on getting it into the ports tree. 

> 
> By the way, can you share how you will be using xnec2c? 

Primarily, for antenna design with 2.4GHZ, 5GHZ, 39GHZ, and LoRa. So much is going on in RF technology right now, it is quite fascinating.

> Are you a ham, and if so, what is your call sign?

No, I wish I was that cool. Although, I have inherited a vintage MARS pin that sits above my monitor, and funny enough, a radio from an Abrams tank is sitting behind me. The ham community is legendary, I probably should invest some time into it someday.

KJ7LNW, 2023-06-09:
> > Thanks for your help! Now xnec2c can claim that it officially supports OpenBSD.
> 
> Now I need to work on getting it into the ports tree.

Awesome!  Let me know if you need anything from me to help with that.

> > By the way, can you share how you will be using xnec2c?
> 
> Primarily, for antenna design with 2.4GHZ, 5GHZ, 39GHZ, and LoRa. So much is going on in RF technology right now, it is quite fascinating.

Cool stuff.  I just looked up LoRa, looks like a neat technology.  I had a colleague doing grad work in the 60GHz range for extremely-short-range highspeed comms.  Short range, because 60GHz is in the oxygen absorption spectrum.

If you're doing antenna design, then you might also take a look at this repository, an addon optimizer for xnec2c: 
 - https://github.com/KJ7LNW/xnec2c-optimize

> > Are you a ham, and if so, what is your call sign?
> 
> No, I wish I was that cool. Although, I have inherited a vintage MARS pin that sits above my monitor, and funny enough, a radio from an Abrams tank is sitting behind me. 

Neat!  

> The ham community is legendary, I probably should invest some time into it someday.

Not sure where you're located, but in the US its just a [35-question test](https://hamstudy.org/tech2022).  If you're already into RF then you probably know much of the test already.  My son Zeke is pretty into it, and he's only 12:
     - [Project Presentation at CadenceLIVE 2023](https://youtu.be/vKeOOaMB4ZQ)
     - [Satellite Tracking Demo](https://youtu.be/vrlw4QPKMRY)

anoduck, 2023-06-15:
@KJ7LNW It sounds to me like Ezekiel has a loving father, and is one very lucky kid. Quite impressive too. 

-----

> Not sure where you're located, but in the US its just a 35-question test. If you're already into RF then you probably know much of the test already.

35 Questions is practically nothing, so I might go for that HAM license when I get a chance.