KJ7LNW/xnec2c #32: Drawing of triangular surface patches

charlieb, 2023-05-08:
The drawing routine for surface patches is only good for rectangular patches and makes triangular patches look really weird and wrong. I'm actually not sure if the underlying nec2 engine handles triangular patches but if it does, I'd be happy to submit a pull request for better drawing of triangular patches.

KJ7LNW, 2023-05-08:
Thank you for offering!

Of course patches are welcome.  Beware, that code is a bit messy:  Refactoring for OpenGL might actually be easier than fixing the old code.  If you're interested in fixing the patch rendering, first please review [#23 (OpenGL Support)](https://github.com/KJ7LNW/xnec2c/issues/23).

If you want to fix existing rendering then go for it, but it is slow software line rendering that redraws every frame and OpenGL really would be best since there is an existing branch with alpha support.

KJ7LNW, 2024-03-31:
Hi @charlieb, would still gladly accept a triangle drawing PR if you're up to it.  If so, then please re-open this issue or post a PR.

closed 2024-03-31 by KJ7LNW