KJ7LNW/xnec2c #36: error building in Lubuntu 16.04

celiponcio, 2023-11-17:
Hello

I'm trying to build xnec2c in Lubuntu 16.04. Could build on other flavor, but I've no indication that it is a flavor issue.
The issue is on ./configure
...
checking whether make sets $(MAKE)... (cached) yes
./configure: line 5353: syntax error near unexpected token `PACKAGE,'
./configure: line 5353: `PKG_CHECK_MODULES(PACKAGE, $pkg_modules)'

tried versions (from repository) 4.4.12, 4.4.11 and 4.3.5.

Also tried to clone the git but get a dependency problem on make:
/usr/lib/x86_64-linux-gnu/glib-2.0/glib-compile-resources --sourcedir=../resources --dependency-file=./.xnec2c-deps/xnec2c-resources.Pc --target=xnec2c-resources.c --generate-source ../resources/xnec2c.gresource.xml
Opção --dependency-file=./.xnec2c-deps/xnec2c-resources.Pc unknown
Makefile:1284: recipe for target 'xnec2c-resources.c' failed

Any advice? Thanks a lot

celiponcio, 2023-11-19:
OK: tried again on a new installation of lubuntu 16.04 and it built from the repository without any problem. Probably it was some issue with previously installed packages.

KJ7LNW, 2023-11-20:
Glad you figured it out!  I think it needed the `autoconf-archive` package.

closed 2023-11-20 by KJ7LNW