KJ7LNW/xnec2c #38: Installation fails

Alain2Bretagne, 2024-03-01:
I tried to install on Ubuntu 22.04.3 LTS with the software manager.  Does not work and no explanation.
I tried from git and I have an error message during the make process:
ake[3] : on quitte le répertoire « /home/alain/Bureau/install xnec2/xnec2c/po »
test ! -f ./xnec2c.pot || \
  test -z "" || make 
touch stamp-po
make[2] : on quitte le répertoire « /home/alain/Bureau/install xnec2/xnec2c/po »
make[2] : on entre dans le répertoire « /home/alain/Bureau/install xnec2/xnec2c »
/usr/bin/sed -e 's|[@]PACKAGE_NAME@|xnec2c|g' -e 's|[@]PACKAGE_STRING@|xnec2c 4.4.12|g' -e 's|[@]PACKAGE_TARNAME@|xnec2c|g' -e 's|[@]PACKAGE_VERSION@|4.4.12|g' < doc/xnec2c.1.in > doc/xnec2c.1
/usr/bin/sed -e 's|[@]PACKAGE_NAME@|xnec2c|g' -e 's|[@]PACKAGE_STRING@|xnec2c 4.4.12|g' -e 's|[@]PACKAGE_TARNAME@|xnec2c|g' -e 's|[@]PACKAGE_VERSION@|4.4.12|g' < xnec2c.spec.in > xnec2c.spec
make[2] : on quitte le répertoire « /home/alain/Bureau/install xnec2/xnec2c »
make[1] : on quitte le répertoire « /home/alain/Bureau/install xnec2/xnec2c »
Making install in src
make[1] : on entre dans le répertoire « /home/alain/Bureau/install xnec2/xnec2c/src »
make[2] : on entre dans le répertoire « /home/alain/Bureau/install xnec2/xnec2c/src »
 /usr/bin/mkdir -p '/usr/local/bin'
  /usr/bin/install -c xnec2c '/usr/local/bin'
/usr/bin/install: impossible de créer le fichier standard '/usr/local/bin/xnec2c': Permission non accordée
make[2]: *** [Makefile:469 : install-binPROGRAMS] Erreur 1
make[2] : on quitte le répertoire « /home/alain/Bureau/install xnec2/xnec2c/src »
make[1]: *** [Makefile:1106 : install-am] Erreur 2
make[1] : on quitte le répertoire « /home/alain/Bureau/install xnec2/xnec2c/src »
make: *** [Makefile:802 : install-recursive] Erreur 1

I did not copy everything, but seems correct before.
As a reseult from th installation, I have the message included.
What shall I do?
Thank you
[image]

KJ7LNW, 2024-03-01:
It looks like you tried to run it as root which broke permissions on ~/.xnec2c/.  There are other issues, too: your install output shows version 4.4.12 but the graphical window shows 4.3.4.

This is what I'd recommend:

 - sudo apt-get remove xnec2c
 - sudo apt-get install build-essential gnome-devel libgtk-3-dev gettext autopoint
 - git clone https://github.com/KJ7LNW/xnec2c.git
 - cd xnec2c
 - ./autogen.sh
 - ./configure
 - make
 - sudo make install
 - sudo make desktop-install # Optional: for icons and file association.
 - sudo chown -R alain.alain /home/alain/.xnec2c

Then run it but _not_ as root:
 - xnec2c

KJ7LNW, 2024-03-01:
If you just want to run it from the software manager, then this should fix your permissions issue:
- sudo chown -R alain.alain /home/alain/.xnec2c

KJ7LNW, 2024-03-01:
You might also need this:

- `sudo chmod 644 /home/alain/.xnec2c/*`

KJ7LNW, 2024-03-04:
Closing this, but re-open if you still have trouble.

KJ7LNW, 2024-03-04:
.

closed 2024-03-04 by KJ7LNW

Alain2Bretagne, 2024-03-04:
Thank you

Sent from [Proton Mail](https://proton.me/mail/home) for iOS