KJ7LNW/xnec2c #39: Installation Issues

nexmug2, 2024-03-07:
Hi Im new to Linux and was trying to install Xnec2c on WSL Ubuntu which I believe should have support. When I had followed the compilation notes I received the following error when inputting the final command 
xnec2c
The error that followed said "Unable to open libtatlas.so.3 : libtatlas.so.3: cannot open a shared object file: No such file or directory skipping." followed by many errors w/ similar messaging but varying in lib_____ etc. 
This puzzled me as I had already downloaded the package libatlas-base-dev, and searched to find if I was missing any additional packages put was unable to find any. Apologies if this is a rookie error but I would like to be able to use Xnec2c.

KJ7LNW, 2024-03-08:
Interesting, I've not tried it with WSL and I don't have a windows system to test with.  Are you building from GIT?  If not, follow these directions:

- sudo apt-get remove xnec2c
- sudo apt-get install git build-essential gnome-devel libgtk-3-dev gettext autopoint
- git clone https://github.com/KJ7LNW/xnec2c.git
- cd xnec2c
- ./autogen.sh
- ./configure
- make && sudo make install
- sudo make desktop-install # Optional: for icons and file association.
- xnec2c

KJ7LNW, 2024-03-31:
Closing issues, but reopen let me know if you have any issues.

closed 2024-03-31 by KJ7LNW