KJ7LNW/xnec2c #44: Command-line interface "batch" option should start the frequency loop automatically

ProtonTheFox, 2024-04-28:
Hi, I'm using Python for automating tasks in the xnec2c workflow. I'm trying to automate the NEC files generation, simulations with xnec2c and optimizations.

During my research on how to perform these things, I found the "batch" option which closes the program after the frequency loop is done, and by adding proper options I can export the results in a Touchstone file. However, the frequency loop doesn't start automatically; I have to start it manually, so it can perform the calculations, write the results in the file, and close the program. That's a quite cumbersome and not practical behaviour.

It would be more relevant that the "batch" option would perform automatically all the stuff without needing a user's action, so it could be effectively used to perform batch simulations.

KJ7LNW, 2024-04-28:
If you load xnec2c and press the triangle "play" button on the freqplots window, and then exit xnec2c, then it should save that setting for next time.

However, you are right: batch should do the right thing.  I'll see what can be done.  Note that --optimize already does something similar, so using --optimize with --batch could be useful.

KJ7LNW, 2024-04-28:
Also, I know you're using Python, but you might take a look at how [xnec2c-optimize](url) uses the `--optimize` feature:
https://github.com/KJ7LNW/xnec2c-optimize

Basically with `--optimize`, xnec2c will auto-reload the .nec file and write the .s2p file without a start/stop cycle.

ProtonTheFox, 2024-04-28:
Thanks for the quick reply.

I already knew about the `--optimize` feature, but apparently it can't be used simultaneously with `--batch`. I get the following message: `[crit] --batch and --optimize are mutual exclusive.` That makes sense though, as `--optimize` implies something iterative, that needs to keep xnec2c opened when calculations are finished, while `--batch` is the exact opposite. But in both cases, it should start automatically the calculations so these modes are useful for automation stuff.

I also already looked how external optimizers such as xnec2c-optimize works, but it's not really what I wanted to do.

> If you load xnec2c and press the triangle "play" button on the freqplots window, and then exit xnec2c, then it should save that setting for next time.

This doesn't work, it keeps on asking me to press the "play" button the next time I call xne2c through my script.

KJ7LNW, 2024-04-28:
You're right, its supposed to do that, so its a bug:

https://github.com/KJ7LNW/xnec2c/blob/54118dd983c714c442ae39867a34975417cddac0/src/main.c#L203-L206

but it doesn't because main_loop_start resets when the config is read, and your ~/.xnec2c/xnec2c.conf has the loop start disabled (it will autosave if you click "play" without batch and then exit):

https://github.com/KJ7LNW/xnec2c/blob/54118dd983c714c442ae39867a34975417cddac0/src/main.c#L381

working on a patch...

KJ7LNW, 2024-04-28:
Rebuild from the master branch and see if that fixes it for you, re-open if there is an issue.

closed 2024-04-28 by KJ7LNW

KJ7LNW, 2024-04-28:
(let me know if it works, too!)

ProtonTheFox, 2024-04-29:
I've just built and tested, and now it works perfectly. Thanks a lot for your quick response and fix!

ProtonTheFox, 2024-04-29:
Looks like I posted a bit too fast, it still doesn't work even if it worked as expected on the first try. I'm investigating it a bit further.

ProtonTheFox, 2024-04-29:
After several tries with various NEC files (mostly the included examples) I can confirm it doesn't work. Which is odd, because it behaved properly the first time I tried the fix!

Here is the command I use to call my freshly built, uninstalled executable:

`./xnec2c -i ../examples/13cm_Yagi.nec --batch --write-s1p ~/test.s1p`

I also tried with `-b` instead of writing it all, no difference. Neither installing xnec2c using `make install`.

I noticed several things that could help however.

First, I tried to delete my `xne2c.conf` file to see if it messed up with the latest version. No success. I inspected the `xne2c.conf` that was then freshly created and did not see any reference to the "batch" mode, while I expected to have something that would be related. Here is the conf file if this helps:
[xnec2c.txt](https://github.com/KJ7LNW/xnec2c/files/15152424/xnec2c.txt)

I also tried to add the `--verbose` option, because from what I understood in the snippet you included, I should have a message saying: `[notice] batch mode enabled, will exit after first loop`. But I didn't get such a message in the console. I got some notice message about gaussian elimination and so on, so I'm sure this command is taken into account. So, maybe something's off in the processing of this option?

(I've got also errors about ATLAS and Gtk, but I don't know if it's related to my installation or a bug. It's probably also another issue anyway, so I won't go further on this subject)

I'll try to check out if I notice something else.

commit 75b439d1 (2024-04-28): bugfix: force main_loop_start=1 with --batch

Thanks to @ProtonTheFox for reporting #44.

Fixes: #44