KJ7LNW/xnec2c #46: Ubuntu 22.04.4 “jammy”

roberthadow, 2024-05-03:
I copied the install script for Xnec2c from the user manual I found at xnec2c.org.  When I ran it, the first hiccough was at gettext.   From there, lots of “no such file” errors.

So I installed the gnome libraries successfully (“sudo apt-get install build-essential gnome-devel libgtk-3-dev gettext autopoint”) and tried the original install script again. 

Then I was told I need GNU Autoconf 2.52.  I installed that. 

Then the install script failed at ATLAS.  With no other help immediately available, I gave up.

I'd like to build antennae.  I need some help on this UNIX stuff. Is there a way to install the app without building it ground up? 73

KJ7LNW, 2024-05-03:
Does this work?  I just tried it here on a minimal installation and it worked:

1. `apt install git build-essential libgtk-3-dev gettext autopoint autoconf libtool gnome-icon-theme desktop-file-utils`
2. `git clone https://github.com/KJ7LNW/xnec2c.git`
3. `cd xnec2c`
4. `./autogen.sh`
5. `./configure`
6. `make`

Then test it:

- `./src/xnec2c examples/airplane.nec `

and install it system-wide
- `make install`

and optionally install desktop components and file associations
- `make desktop-install`

roberthadow, 2024-05-03:
Hello KJ7LNW --

We can put closed to the Github ticket.  I altered the script and 
subsequent commands to include sudo, as is my practice.  It all worked 
without a hitch.

Many thanks,

73

W2NJ

roberthadow, 2024-05-04:
I'd recommend updating the recommended script at xnec2c.org.  The one provided to me above worked perfectly. Thanks

closed 2024-05-04 by roberthadow

KJ7LNW, 2024-05-04:
> I'd recommend updating the recommended script at xnec2c.org. The one provided to me above worked perfectly. Thanks

You're welcome. We updated documentation for Ubuntu on the website accordingly.