KJ7LNW/xnec2c #47: Segment tag number error

roberthadow, 2024-05-04:
Update docs with troubleshooting hints for "Segment tag number error."

While the underlying user error may be common to all NEC2 applications, the error message appears to be unique.  A sentence or two in the user manual is easy; a hint in the application error message might be more work.

In the mean time, the user is prevented from saving his or her work, which may be substantial.

KJ7LNW, 2024-05-04:
Can you provide the .NEC file which generated this error?

KJ7LNW, 2024-05-04:
> In the mean time, the user is prevented from saving his or her work, which may be substantial.

Calculations cannot run unless the .NEC file has been saved. Thus, as far as I can tell, it is not possible to get this error unless the file has been saved. If you believe this assessment is incorrect, then please provide a step-by-step procedure to generate the error in such way that the file on disk differs from the file in the editor.

roberthadow, 2024-05-04:
This file, generated within the application, produces "Segment tag number error": Moxon240503.nec. [attached with ".txt" extension.  Apparently, it did save, even though when I clicked the Save icon, I got the error, implying that the command had not completed.  If I had looked at the file characteristics, I would have seen that the OS knew it had been recently modified.

Another file, created outside the application, produces "Segment data error, zero-length line? See console."  Moxon240504.nec

If we can capture this class of issue--stupid user tricks and obscure error messages--we can add text to the user manual written once/read many until the app or the error message changes.  This is work I for which I would respond to a Help Wanted ad.

[Moxon240504.nec.txt](https://github.com/KJ7LNW/xnec2c/files/15211154/Moxon240504.nec.txt)
[Moxon240503.nec.txt](https://github.com/KJ7LNW/xnec2c/files/15211155/Moxon240503.nec.txt)

KJ7LNW, 2024-05-05:
Unfortunately NEC2 tries to parse your input _and then_ calculates segment sizes, so it cannot always produce a meaningful error. If you run `xnec2c` from the command line it may provide more information about the error that is not show in the pop-up window (like tag number).

- In [Moxon240503.nec.txt](https://github.com/KJ7LNW/xnec2c/files/15211155/Moxon240503.nec.txt), your `EX` tag tels it to excite segment-10 of tag-1, but tag-1 only has one segment defined:
   - GA     1     **1** ...  
   - EX     0     1     **10** ...

Additionally, you have two excitations (`EX`) defined. I believe that this is acceptable, for example you might have expectations on multiple segments at different phases in some antenna geometries. However, if you are just getting started that I recommend sticking to a single excitation line. The console output shows the following:

```
[err] no segment has an itag of 1
[err] Stop: Segment tag number error
```

- In [Moxon240504.nec.txt](https://github.com/KJ7LNW/xnec2c/files/15211154/Moxon240504.nec.txt),  your GS tag is all zeros, so it scales the GA tag-1 line to nothing, thus creating a zero length line segment(s).   The console output shows the following:
```
[err] segment data error: tag=1 si=0.000000, bi=0.000000
[err] Stop: Segment data error, zero-length line?  See console.
```

The official an EC to documentation is available here:
- http://www.nec2.org/

These are the references that I used to diagnose the errors that your geometry generated:

- http://www.nec2.org/part_3/cards/ga.html
- http://www.nec2.org/part_3/cards/gs.html
- http://www.nec2.org/part_3/cards/ex.html

You are correct in that the graphical error could be more meaningful by also displaying the console errors in graphical output.  If the stop function was read factored to support variable length arguments, then cancel warnings could be merged into those warning outputs and be slightly more verbose are the graphical interface. If you have some background in see programming then it would be pretty easy to convert `Stop()`  into something that supports variable length arguments:

https://github.com/KJ7LNW/xnec2c/blob/994927d50effc745924c15bd259989700c71df22/src/utils.c#L91-L93

I opened issue #48  if that is something that you would like to tackle with a pull request.

closed 2024-06-09 by KJ7LNW