KJ7LNW/xnec2c #49: User experience: Window titles

roberthadow, 2024-05-05:
Choose and use consistently two short names for the Xnec2c windows.  "Main" and "Editor" are two possibilities. "Xnec2c Console" and "Xnec2c Editor" are also possibilities.

Display the file name or first line of comment (CM) in the window title bar for both windows.  If the string is too long, devise a clever means to truncate it.

This will help the (new) user relate the documentation to the application.

KJ7LNW, 2024-05-06:
> Choose and use consistently two short names for the Xnec2c windows. "Main" and "Editor" are two possibilities. "Xnec2c Console" and "Xnec2c Editor" are also possibilities.

What do you mean?

> Display the file name or first line of comment (CM) in the window title bar for both windows. If the string is too long, devise a clever means to truncate it.

Good idea.  Do you know how to set that?

KJ7LNW, 2025-12-19:
Both of your training requests have been replied to. Please build from the latest skit tip and give it a shot.

closed 2025-12-19 by KJ7LNW