KJ7LNW/xnec2c #50: xnec2c still beeps

polprog, 2024-05-12:
Hi Eric,

First of all thank you for developing this great program! I finally found a modern NEC implementation that works on Linux.

I can see there is a commit from March 31 which fixes some of the beeps
https://github.com/KJ7LNW/xnec2c/commit/7bba2b40b9ccaf27f7367d61c48588a141e0a1be

however I built [Release v4.4.16](https://github.com/KJ7LNW/xnec2c/releases/tag/v4.4.16), and it still keeps beeping when rotating the 3d views.

Is this some GTK "feature" or is it possible to disable it?

KJ7LNW, 2024-05-13:
> First of all thank you for developing this great program! I finally found a modern NEC implementation that works on Linux.

I am glad that you like it, however, I should note that this was actually written by [Neoklis Kyriazis 5B4AZ](http://www.5b4az.org/). I am just now the maintainer for features and bug fixes because since Neoklis handed it off me to maintain.

> I can see there is a commit from March 31 which fixes some of the beeps [7bba2b4](https://github.com/KJ7LNW/xnec2c/commit/7bba2b40b9ccaf27f7367d61c48588a141e0a1be)
> 
> however I built [Release v4.4.16](https://github.com/KJ7LNW/xnec2c/releases/tag/v4.4.16), and it still keeps beeping when rotating the 3d views.
> 
> Is this some GTK "feature" or is it possible to disable it?

I do not know if the problem you are experiencing is the same or similar to the one in the commit that your referenced, but if it is, it is caused by a digit overflow in on of the information fields that the top of the window. 

Can you send me the .NEC file which causes the issue, so I can try to reproduce it on my system?

polprog, 2024-05-13:
I managed to reproduce it on the default starting dipole file (the one that is loaded with File > New). I noticed it happens when looking at the antenna parallel to the Z axis, which makes sense since the gain is listed as -999,9 dB. 

[untitled.nec.txt](https://github.com/KJ7LNW/xnec2c/files/15299480/untitled.nec.txt)

This is the file that initially beeped a lot on 4.4.12 but on 4.4.16 beeps only when looking at the antenna from the y axis where there is -inf dB gain as well:
[yagi14MHz.nec.txt](https://github.com/KJ7LNW/xnec2c/files/15299609/yagi14MHz.nec.txt)

KJ7LNW, 2024-05-13:
in `resources/xnec2c.glade`:

Try chaning this `6` to `9` on line 11302:

https://github.com/KJ7LNW/xnec2c/blob/94c96d7f416883d6574c3385f4f6c4ad1f59d904/resources/xnec2c.glade#L11302

and also here, change `6` to `9` on line 12586:

https://github.com/KJ7LNW/xnec2c/blob/94c96d7f416883d6574c3385f4f6c4ad1f59d904/resources/xnec2c.glade#L12586

then `make clean && make` and try again.  If that fixes it I'll do a release...

KJ7LNW, 2024-05-23:
@polprog I think this is fixed, please test a965e7f in git and let me know if it still beeps.

commit a965e7f6 (2024-05-23): prevent more gtk beeps

Thanks to @polprog for reporting #50, which appears to be more GTK beeps
related to the ones fixed in 7bba2b4 .

Fixes: #50
closed 2024-05-23 by KJ7LNW