KJ7LNW/xnec2c #58: Memory Re-Allocation Denied

DonLocke, 2024-11-25:
When running xnec2c with the xnec2c-gao optimizer I regularly get the error `Memory re-allocation denied in geometry.c` which crashes the application and I unfortunately loose all progress that gao has made.

I don't believe this is a problem with xnec2c-gao, it is simply running the command:
`xnec2c  --optimize -j7  -i  [myFile].nec > /dev/null 2>&1`

I'm running xnec2c 4.4.16, installed on Manjaro (Arch-based) linux. I'm not sure where else to look to get better logs to debug the issue.

Thank you for any help you can provide!

DonLocke, 2024-11-26:
After some more testing, I was able to find get a card that always causes the issue, I'm not sure why these specific GW entries cause it to fail though, nothing seems obviously incorrect to me. It's specifically the combination of the two GWs, individually they load fine.
```
CM 	--- NEC2 Input File created or edited by xnec2c 4.4.16 ---
CM 	Memory re-allocation denied
CE 	--- End Comments ---
GW 	        1	       25	0,3136937	0,42493248	      0,0	0,3136937	-0,42493248	      0,0	   1,6e-3
GW 	        2	       20	0,3131709	0,42493248	      0,0	0,3136937	0,42493248	      0,0	   1,6e-3
GE 	0	0	0,00000E+00	0,00000E+00	0,00000E+00	0,00000E+00	0,00000E+00	0,00000E+00	0,00000E+00
EX 	0	1	13	0	1,00000E+00	0,00000E+00	0,00000E+00	0,00000E+00	0,00000E+00	0,00000E+00
LD 	5	0	0	0	3,70000E+07	0,00000E+00	0,00000E+00	0,00000E+00	0,00000E+00	0,00000E+00
NH 	0	0	0	0	0,00000E+00	0,00000E+00	0,00000E+00	0,00000E+00	0,00000E+00	0,00000E+00
NE 	0	20	15	1	-1,40000E+00	-1,40000E+00	5,00000E-02	2,00000E-01	2,00000E-01	0,00000E+00
RP 	0	37	73	1000	0,00000E+00	0,00000E+00	5,00000E+00	5,00000E+00	0,00000E+00	0,00000E+00
FR 	        0	       25	0	0	    144,0	   8,0e-2	    146,0
EN
```

DonLocke, 2024-11-26:
> After some more testing, I was able to find get a card that always causes the issue, I'm not sure why these specific GW entries cause it to fail though, nothing seems obviously incorrect to me. It's [...]

It also only starts to fail when GW 2 Segments are >=13, if that helps

KJ7LNW, 2024-11-26:
I have not yet looked in detail, but typically these crashes are related to GW line intersections that do not me at a segment vertex, or lines that overlap at least partially. Is this possible?

KJ7LNW, 2024-12-29:
fyi, crash was "fixed" via work-around in unreleased b085ded2 . The real issue is segment distance problems.

These warnings are not yet committed (soon!), but I have tested them on the file that you gave me in this is what I see:

[warn] Tag 2 segment 44 length 2.614e-05 is too short compared to tag 1 segment 1 length 3.399e-02; These segments are close enough that connection detection issues may occur
[warn] Tag 2 segment 45 length 2.614e-05 is too short compared to tag 1 segment 1 length 3.399e-02; These segments are close enough that connection detection issues may occur

[warn] tag=2/seg=44: length=2.614e-05 too short (and close enough) compared to tag=1/seg=1 length=3.399e-02; connection detection may fail
[warn] tag=2/seg=45: length=2.614e-05 too short (and close enough) compared to tag=1/seg=1 length=3.399e-02; connection detection may fail
[err] tag=2/seg=26: length/radius ratio of 0.0 < 2.0; standard thin-wire kernel field error exceeds 1% by a large margin (min=2.0, 1%accurate=8.0)
[err] tag=2/seg=27: length/radius ratio of 0.0 < 2.0; standard thin-wire kernel field error exceeds 1% by a large margin (min=2.0, 1%accurate=8.0)
[err] tag=2/seg=28: length/radius ratio of 0.0 < 2.0; standard thin-wire kernel field error exceeds 1% by a large margin (min=2.0, 1%accurate=8.0)
[err] tag=2/seg=29: length/radius ratio of 0.0 < 2.0; standard thin-wire kernel field error exceeds 1% by a large margin (min=2.0, 1%accurate=8.0)
[err] tag=2/seg=30: length/radius ratio of 0.0 < 2.0; standard thin-wire kernel field error exceeds 1% by a large margin (min=2.0, 1%accurate=8.0)
[err] tag=2/seg=31: length/radius ratio of 0.0 < 2.0; standard thin-wire kernel field error exceeds 1% by a large margin (min=2.0, 1%accurate=8.0)
[err] tag=2/seg=32: length/radius ratio of 0.0 < 2.0; standard thin-wire kernel field error exceeds 1% by a large margin (min=2.0, 1%accurate=8.0)
[err] tag=2/seg=33: length/radius ratio of 0.0 < 2.0; standard thin-wire kernel field error exceeds 1% by a large margin (min=2.0, 1%accurate=8.0)
[err] tag=2/seg=34: length/radius ratio of 0.0 < 2.0; standard thin-wire kernel field error exceeds 1% by a large margin (min=2.0, 1%accurate=8.0)
[err] tag=2/seg=35: length/radius ratio of 0.0 < 2.0; standard thin-wire kernel field error exceeds 1% by a large margin (min=2.0, 1%accurate=8.0)
[err] tag=2/seg=36: length/radius ratio of 0.0 < 2.0; standard thin-wire kernel field error exceeds 1% by a large margin (min=2.0, 1%accurate=8.0)
[err] tag=2/seg=37: length/radius ratio of 0.0 < 2.0; standard thin-wire kernel field error exceeds 1% by a large margin (min=2.0, 1%accurate=8.0)
[err] tag=2/seg=38: length/radius ratio of 0.0 < 2.0; standard thin-wire kernel field error exceeds 1% by a large margin (min=2.0, 1%accurate=8.0)

KJ7LNW, 2024-12-29:
Please build the [segment-tests](https://github.com/KJ7LNW/xnec2c/tree/segment-tests) branch if you would like to test the additional verifications.