KJ7LNW/xnec2c #59: The saved NEC file is invalid for some locales

hamarituc, 2024-12-15:
The saved NEC file depends on the system locale. The number format produces invalid NEC files when a NEC deck is saved by `xnec2c`.

1. Load the following.

```
CM --- NEC2 Input File created or edited by xnec2c 4.4.12 ---
CE --- End Comments ---
GW     1   101  -1.00000E+01  0.00000E+00  0.00000E+00  1.00000E+01  0.00000E+00  0.00000E+00  1.00000E-03
GE     0     0   0.00000E+00  0.00000E+00  0.00000E+00  0.00000E+00  0.00000E+00  0.00000E+00  0.00000E+00
EX     0     1    51      0  1.00000E+00  0.00000E+00  0.00000E+00  0.00000E+00  0.00000E+00  0.00000E+00
FR     0     1     0      0  0.70000E+01  1.00000E-01  1.00000E+02  0.00000E+00  0.00000E+00  0.00000E+00
RP     0    37    73   1000  0.00000E+00  0.00000E+00  5.00000E+00  5.00000E+00  0.00000E+00  0.00000E+00
NE     0    10     4      4 -1.80000E+01 -0.60000E+01 -0.60000E+01  4.00000E+00  4.00000E+00  4.00000E+00
NH     0    10     4      4 -1.80000E+01 -0.60000E+01 -0.60000E+01  4.00000E+00  4.00000E+00  4.00000E+00
EN     0     0     0      0  0.00000E+00  0.00000E+00  0.00000E+00  0.00000E+00  0.00000E+00  0.00000E+00
```

2. Execute `xnec2` with a german locale (for example).

```
LC_NUMERIC="de_DE.UTF-8" xnec2c
```

3. Load the file from above
4. Save the file produces (a comma `,` is used as decimal separator instead of a dot `.`)

```
CM --- NEC2 Input File created or edited by xnec2c 4.4.12 ---
CE --- End Comments ---
GW     1   101  -1,00000E+01  0,00000E+00  0,00000E+00  1,00000E+01  0,00000E+00  0,00000E+00  1,00000E-03
GE     0     0   0,00000E+00  0,00000E+00  0,00000E+00  0,00000E+00  0,00000E+00  0,00000E+00  0,00000E+00
EX     0     1    51      0  1,00000E+00  0,00000E+00  0,00000E+00  0,00000E+00  0,00000E+00  0,00000E+00
FR     0     1     0      0  7,00000E+00  1,00000E-01  1,00000E+02  0,00000E+00  0,00000E+00  0,00000E+00
RP     0    37    73   1000  0,00000E+00  0,00000E+00  5,00000E+00  5,00000E+00  0,00000E+00  0,00000E+00
NE     0    10     4      4 -1,80000E+01 -6,00000E+00 -6,00000E+00  4,00000E+00  4,00000E+00  4,00000E+00
NH     0    10     4      4 -1,80000E+01 -6,00000E+00 -6,00000E+00  4,00000E+00  4,00000E+00  4,00000E+00
EN     0     0     0      0  0,00000E+00  0,00000E+00  0,00000E+00  0,00000E+00  0,00000E+00  0,00000E+00
```

This file cannot be parsed successfully by `nec2c` for example. The output format shouldn't depend on the locale.

KJ7LNW, 2024-12-17:
You managed to find a bug that was very difficult. I might have found a simple hackfix related to saving, but the proper fix was to convert all string representations of fields in the editor to ints and doubles. 

Because this change is so large, there is a risk of error in the implementation. If you are familiar with programming, then I would greatly appreciate your review to see if I missed anything during the data-type refactor.

Before I do a release, please test as follows and see if this fixes your problem. If you can test the graphical NEC2 file editor interface extensively to see if there are any unexpected side effects, then I would appreciate that as well.

- git clone https://github.com/KJ7LNW/xnec2c.git
- cd xnec2c
- git checkout origin/bug-59
- ./autogen.sh
- ./configure
- make 
- ./src/xnec2c <your-nec-file> # this runs from build tree without install

hamarituc, 2024-12-19:
> You managed to find a bug that was very difficult. I might have found a simple hackfix related to saving, but the proper fix was to convert all string representations of fields in the editor to ints and doubles.
> 
> Because this change is so large, there is a risk of error in the implementation. If you are familiar with programming, then I would greatly appreciate your review to see if I missed anything during the data-type refactor.
> 
> Before I do a release, please test as follows and see if this fixes your problem. If you can test the graphical NEC2 file editor interface extensively to see if there are any unexpected side effects, then I would appreciate that as well.
> 
>     * git clone https://github.com/KJ7LNW/xnec2c.git
> 
>     * cd xnec2c
> 
>     * git checkout origin/bug-59
> 
>     * ./autogen.sh
> 
>     * ./configure
> 
>     * make
> 
>     * ./src/xnec2c  # this runs from build tree without install

I am happy to confirm the patch works for me. Thanks for the fast fix.

FYI: According to the man-page I would assume that `setlocale(LC_NUMERIC, "");` should also work to reset to the default locale and it is not necessary to save the original locale.

```diff
diff --git a/src/nec2_model.c b/src/nec2_model.c
index 2255d66..8237b2d 100644
@@ -648,6 +716,7 @@ Save_Nec2_Input_File( GtkWidget *treeview_window, char *nec2_file )
 {
   FILE *nec2_fp = NULL;
 
+  setlocale(LC_NUMERIC, "C");
   /* Abort if editor window is not opened */
   if( nec2_edit_window == NULL ) return;
 
@@ -667,6 +736,8 @@ Save_Nec2_Input_File( GtkWidget *treeview_window, char *nec2_file )
   /* Close file to re-open in read mode */
   Close_File( &nec2_fp );
 
+  setlocale(LC_NUMERIC, orig_numeric_locale);
+
   ClearFlag( NEC2_SAVE );
   ClearFlag( NEC2_EDIT_SAVE );
 
```

KJ7LNW, 2024-12-19:
> I am happy to confirm the patch works for me. Thanks for the fast fix.

Excellent!  

> FYI: According to the man-page I would assume that `setlocale(LC_NUMERIC, "");` should also work to reset to the default locale and it is not necessary to save the original locale.

Interesting, good to know!

closed 2024-12-29 by KJ7LNW

KJ7LNW, 2024-12-29:
I merged to the main branch and closed the ticket, so reopen if there is a problem.

commit dd5e9659 (2024-12-29): fix: Ensure NEC file format uses decimal points regardless of locale

The NEC file format requires decimal points (.) for floating point numbers
but the output was using the system locale's decimal separator (e.g. comma
in German locale). This caused invalid NEC files to be generated when
saving from xnec2c.

- Store numeric values as native types (int/double) in GTK list store
- Format numbers with decimal point when saving to file by using C locale
- Restore original locale after file save to maintain UI consistency
- Refactor data handling to avoid string/number conversions
- Fix format tru [... cut]
commit 5466682c (2024-12-29): fix: Ensure NEC file format uses decimal points regardless of locale

The NEC file format requires decimal points (.) for floating point numbers
but the output was using the system locale's decimal separator (e.g. comma
in German locale). This caused invalid NEC files to be generated when
saving from xnec2c.

- Store numeric values as native types (int/double) in GTK list store
- Format numbers with decimal point when saving to file by using C locale
- Restore original locale after file save to maintain UI consistency
- Refactor data handling to avoid string/number conversions
- Fix format tru [... cut]