KJ7LNW/xnec2c #60: Feature Request - Graphical Editor

ds2k5, 2024-12-21:
Hi,
is it possible to add a Graphical Editor like 4nec2 has ?

Example Picture:  https://image.slideserve.com/437737/4nec2-graphical-editor-for-beginners-l.jpg

I am not a programmer so I can not build it

KJ7LNW, 2024-12-23:
Of course, it is possible, however I also believe it would be a large development effort. Patches are welcome if someone would like to add this feature. 

Actually xnec2c does have a "graphical" editor of sorts, but it does not automatically save and render unless you click save. I have often thought it would be neat to automatically save every time you make any change, so you can see a live view of the model as you modify, but I have not looked into that yet. 

By the way, 4nec2 works in [WINE](https://www.winehq.org/), and xnec2c has an [optimization feature](https://www.xnec2c.org/#Optimizers) that will reload render every time the file is saved. This means you could edit in Linux using the WINE-emulated version of 4nec2, and then every time you save, xnec2c will render it nice and fast in parallel (with -j `nproc`) if you enable File->Optimize.

See also: 
- https://github.com/KJ7LNW/xnec2c-optimize/

ds2k5, 2024-12-23:
Thank you

>By the way, 4nec2 works in [WINE](https://www.winehq.org/),

Yes but the developer will not offer the source
and then the project is in "danger" if the developer did not have time to Maintain the code