KJ7LNW/xnec2c #61: Display/Save currents

g4ixt, 2024-12-28:
xnec2c v4.4.16 on Kubuntu 24.04.  14MHz folded dipole, centre excited with 77V (P=20W) set with Z0 = 300 Ohm, resonant on 14.239MHz

When I run the model and select "view currents," the graduated colour scale is missing (bug?) but at the resonant frequency it shows the max current as 2.59E-1 Amps on screen.  Plugging this back into the usual formula with P gives R of 298 Ohms as expected.

When I select a frequency of 14.259MHz in the "view currents" window and then File > save currents/charges as CSV, I have a file with the correct frequency in column A but for no tag/segment do I see a current of 2.59E-1 Amps, they are all very much smaller. 

I may very well be misunderstanding or doing something wrong.  Files attached.  Any advice?

Ian G4IXT

(Supplementary Question: What units are "charges" in and is there any way to relate them to Voltage on the wire?)
[g4ixt_folded_v8_14MHz.tar.gz](https://github.com/user-attachments/files/18266209/g4ixt_folded_v8_14MHz.tar.gz)

KJ7LNW, 2024-12-28:
> When I run the model and select "view currents," the graduated colour scale is missing (bug?) 

that is strange the color scale works on my screen, so maybe it is a distribution issue? but let's get to the math problem:

> but at the resonant frequency it shows the max current as 2.59E-1 Amps on screen. Plugging this back into the usual formula with P gives R of 298 Ohms as expected.
>...
> have a file with the correct frequency in column A but for no tag/segment do I see a current of 2.59E-1 Amps, they are all very much smaller.

In Draw_Wire_Segments(), when displaying currents, the code shows:

```c
if( isFlagSet(DRAW_CURRENTS) )
  snprintf( label, s, "%8.2E", cmax * (double)data.wlam );
```

The displayed current is scaled by wavelength (data.wlam), but in Save_Currents_CSV() the raw current values are saved without this scaling.

So, `cabs(real+imag*i)*lambda` will probably fix it?

Can you confirm that that works for you in the spreadsheet?

If this looks right to you then I will update the function to include a magnitude column scaled by wavelength, but I have another question:

Do you believe that the complex and real value should also be scaled by wave length when writing to the CSV? I am not so familiar with the math here (as I am just the maintainer), so your help would be appreciated.

@neoklis, can you comment on this?

KJ7LNW, 2024-12-28:
(FYI, CSV current/charge save feature was added in c0fc2c879e4f5905238a21f4162b0c4b86e209c2 for #34)

KJ7LNW, 2024-12-28:
see attached does this look correct?

[g4ixt_folded_v8_14MHz-currents2.csv](https://github.com/user-attachments/files/18267974/g4ixt_folded_v8_14MHz-currents2.csv)

KJ7LNW, 2024-12-29:
I believe this is complete and solved, but reopen the issue if the problem persists.

KJ7LNW, 2024-12-29:
and I forgot to mention, if you build this from git then you will have the latest version that contains the fix:

- git clone https://github.com/KJ7LNW/xnec2c.git
- cd xnec2c
- ./autogen.sh
- ./configure
- make
- ./src/xnec2c # test from source tree w/o install

### optional:
- apt remove xnec2c
-  make install
- make desktop-install # Optional: for icons and file association.
- xnec2c # installed from git

closed 2024-12-29 by KJ7LNW

neoklis, 2024-12-29:
Hi Eric!

Best season's greetings to you and Zeke!

>> When I run the model and select "view currents," the graduated colour scale is missing (bug?)

I don't know about this. I think it may happen if min and max values are the same, if possible. 
But I have never experienced this.
 
> So, cabs(real+imag*i)*lambda will probably fix it?

I think so too, did you get the right results in the CSV?

> Do you believe that the complex and real value should also be scaled by wave length when 
> writing to the CSV? I am not so familiar with the math here (as I am just the maintainer),
> so your help would be appreciated.
I think they should be, the absolute scalar value cmag is calculated from them and if you want
to print the real and complex parts of it, they should be scaled.

--
Best Regards

Neoklis - Ham Radio Call: 5B4AZ
http://www.5b4az.org/
https://www.qsl.net/5b4az/index.html

KJ7LNW, 2024-12-29:
Awesome, thanks!  See 01c9bef for the diff.  This fixed both current and charge scaling and added magnitude columns.

g4ixt, 2024-12-29:
> see attached does this look correct?
> 
> [g4ixt_folded_v8_14MHz-currents2.csv](https://github.com/user-attachments/files/18267974/g4ixt_folded_v8_14MHz-currents2.csv)

Yes, that looks correct, thank you.

Ian

g4ixt, 2024-12-29:
> > When I run the model and select "view currents," the graduated colour scale is missing (bug?)
> that is strange the color scale works on my screen, so maybe it is a distribution issue? 

It turns out to be a window size issue.  Since it worked ok for you I thought to try changing the window size (I had 3 windows 'docked' on my widescreen monitor).  As soon as I changed the size, the graduated colour scale appeared.

commit 01c9bef0 (2024-12-29): gnuplot: add current magnitude and scale currents/charges in CSV output

The Save_Currents_CSV function was missing current magnitude output and
proper scaling of currents and charges. This commit:

- Adds current_mag column to match the on-screen display
- Scales currents by wavelength (data.wlam) to match display values
- Scales charges by 1.0E-6/freq_mhz for proper units (Coulombs/m)
- Updates CSV header to include new magnitude columns:
  * current_real, current_imag, current_mag
  * charge_real, charge_imag, charge_mag

Thanks to Ian G4IXT for reporting the discrepancy between displayed
a [... cut]