KJ7LNW/xnec2c #62: Frequency loop not ran with --optimize and a single frequency specified

malcolmboyd, 2025-02-23:
Hello,

I have not been able to get the frequency loop to run with the `--optimize` flag on and when specifying a single frequency. I do get a radiation pattern output when I manually open the file in the editor and click save though this doesn't appear to output the csv file required for optimizing. 

Command to open xnec2c:
```bash 
xnec2c -j 16 --optimize -i output/antenna.nec --write-csv output/antenna.csv
``` 

Input file (output/antenna.nec):
```
CM Curtain antenna
CE 
GW     1    31   0.00000E+00  0.00000E+00  9.91943E+00  1.70047E+01  0.00000E+00  9.91943E+00  1.29000E-03
GW     2    31   2.26730E+01  0.00000E+00  9.91943E+00  3.96777E+01  0.00000E+00  9.91943E+00  1.29000E-03
GW     3    31   0.00000E+00  0.00000E+00  1.98389E+01  1.70047E+01  0.00000E+00  1.98389E+01  1.29000E-03
GW     4    31   2.26730E+01  0.00000E+00  1.98389E+01  3.96777E+01  0.00000E+00  1.98389E+01  1.29000E-03
GW     5    32  -7.93554E-01 -4.67630E+00  2.40900E+01  4.04713E+01 -4.67630E+00  2.40900E+01  1.29000E-03
GW     6    32  -7.93554E-01 -4.67630E+00  2.26730E+01  4.04713E+01 -4.67630E+00  2.26730E+01  1.29000E-03
GW     7    32  -7.93554E-01 -4.67630E+00  2.12559E+01  4.04713E+01 -4.67630E+00  2.12559E+01  1.29000E-03
GW     8    32  -7.93554E-01 -4.67630E+00  1.98389E+01  4.04713E+01 -4.67630E+00  1.98389E+01  1.29000E-03
GW     9    32  -7.93554E-01 -4.67630E+00  1.84218E+01  4.04713E+01 -4.67630E+00  1.84218E+01  1.29000E-03
GW    10    32  -7.93554E-01 -4.67630E+00  1.70047E+01  4.04713E+01 -4.67630E+00  1.70047E+01  1.29000E-03
GW    11    32  -7.93554E-01 -4.67630E+00  1.55877E+01  4.04713E+01 -4.67630E+00  1.55877E+01  1.29000E-03
GW    12    32  -7.93554E-01 -4.67630E+00  1.41706E+01  4.04713E+01 -4.67630E+00  1.41706E+01  1.29000E-03
GW    13    32  -7.93554E-01 -4.67630E+00  1.27536E+01  4.04713E+01 -4.67630E+00  1.27536E+01  1.29000E-03
GW    14    32  -7.93554E-01 -4.67630E+00  1.13365E+01  4.04713E+01 -4.67630E+00  1.13365E+01  1.29000E-03
GW    15    32  -7.93554E-01 -4.67630E+00  9.91943E+00  4.04713E+01 -4.67630E+00  9.91943E+00  1.29000E-03
GW    16    32  -7.93554E-01 -4.67630E+00  8.50237E+00  4.04713E+01 -4.67630E+00  8.50237E+00  1.29000E-03
GW    17    32  -7.93554E-01 -4.67630E+00  7.08531E+00  4.04713E+01 -4.67630E+00  7.08531E+00  1.29000E-03
GW    18    32  -7.93554E-01 -4.67630E+00  5.66824E+00  4.04713E+01 -4.67630E+00  5.66824E+00  1.29000E-03
GW    19    32  -7.93554E-01 -4.67630E+00  4.25118E+00  4.04713E+01 -4.67630E+00  4.25118E+00  1.29000E-03
GE     1     0   0.00000E+00  0.00000E+00  0.00000E+00  0.00000E+00  0.00000E+00  0.00000E+00  0.00000E+00
FR     0     1     0     0  1.410000E+00  0.00000E+00  0.00000E+00  0.00000E+00  0.00000E+00  0.00000E+00
GN     2     0     0      0  2.00000E+01  3.03000E-02  0.00000E+00  0.00000E+00  0.00000E+00  0.00000E+00
EX     0     1    16      0  1.00000E+00  0.00000E+00  0.00000E+00  0.00000E+00  0.00000E+00  0.00000E+00
EX     0     2    16      0  1.00000E+00  0.00000E+00  0.00000E+00  0.00000E+00  0.00000E+00  0.00000E+00
EX     0     3    16      0  1.00000E+00  0.00000E+00  0.00000E+00  0.00000E+00  0.00000E+00  0.00000E+00
EX     0     4    16      0  1.00000E+00  0.00000E+00  0.00000E+00  0.00000E+00  0.00000E+00  0.00000E+00
RP     0    45    60   1000  9.00000E+01  0.00000E+00 -2.00000E+00  6.00000E+00  0.00000E+00  0.00000E+00
EN     0     0     0      0  0.00000E+00  0.00000E+00  0.00000E+00  0.00000E+00  0.00000E+00  0.00000E+00
```

KJ7LNW, 2025-02-24:
Interesting, does not work with more than one frequency?

if so then there is some kind of edge case that is missing for optimization ...

KJ7LNW, 2025-12-19:
The user interface has been fixed up as well for your single-point plot, and it gives frequencies nearby so you can manually select and see how it looks close to your set or sample points. 

[image]