KJ7LNW/xnec2c #63: Radiation pattern gain display

g4ixt, 2025-03-07:
I physically measured my old 23-ele 23cm band Tonna yagi using a vernier caliper and then simulated it, and a pair of them, in Xnec2c.
The gain pattern display is nice to look at but not very useful without a scale display.  I set the yagi along the Y axis and the only way that I can see to determine the side lobe levels is to view "Y" and rotate it about Z whilst looking at the "gain" box, or export the data to a spreadsheet and plot it as a graph.
  
Would it be possible to add a scale to the display, for example as per the attached and/or as the ARRL radiation pattern chart?
  
[Tonna_23el_doc_220623f.pdf](https://github.com/user-attachments/files/19124529/Tonna_23el_doc_220623f.pdf)
  
A second issue is that the gain in my model (which may be an error in my NEC control cards)  and colour scale lower limit is -999.99dB at zero degrees and the colour range is based on this.  I suggest limiting the maximum range to something like 40dB to 60dB, with an on/off limit-checkbox.  This would make the colours more meaningful if it's not possible to add scale lines.  It would be even better if there was a way to determine the actual value from the colour.

I can provide my .nec files if that will help illustrate the issue.

Regards

Ian G4IXT

KJ7LNW, 2025-03-07:
Does `View->Gain Scale` in the radiation pattern window do what you want?

[image]

> set the yagi along the Y axis and the only way that I can see to determine the side lobe levels is to view "Y" and rotate it about Z whilst looking at the "gain" box, or export the data to a spreadsheet and plot it as a graph.

Do the X/Y/Z buttons not accomplish what you are looking for?

You could always use a `GM` card to rotate the whole structure into a plane that works for you.

> I suggest limiting the maximum range to something like 40dB to 60dB, with an on/off limit-checkbox

Neat idea. Clamping is already performed internally at -999. I gave it a try, but I do not see a difference:

**current -999 clamp**
[image]

**proposed -40 clamp**
[image]

g4ixt, 2025-03-07:
> Does `View->Gain Scale` in the radiation pattern window do what you want?
  
No, because there are no reference scale lines on the display and that just changes the actual scaling from linear to log or ARRL formula.  

> Do the X/Y/Z buttons not accomplish what you are looking for?
  
No all they do is to show the pattern from different views

> You could always use a `GM` card to rotate the whole structure into a plane that works for you.
  
I can already rotate it using the spinboxes and that is the only way to see the actual 'dB' values of side lobes unless I export the data and plot it on a graph.
  
> > I suggest limiting the maximum range to something like 40dB to 60dB, with an on/off limit-checkbox 
> Neat idea. Clamping is already performed internally at -999. I gave it a try, but I do not see a difference:
  
It looks the same, that's not what I expected.  Assuming the colours change on their "x-axis" in a "linear" way, I would say that the top one "green" starts at about -500dB and on the bottom one it starts at about -17dB so I would expect the colours on the main lobe to be different in the second one.  -999dB is essentially meaningless when a range of just 60dB represents a power ratio of a million to one.  Unless you had access to an an-echoic chamber for RF you wouldn't be able to measure an antenna pattern to that precision so there isn't much point modelling it beyond say 70dB  - although I am not an expert in this field.

g4ixt, 2025-03-07:
[image]

e.g. if the red is +19.38dB, how do I tell what blue means

g4ixt, 2025-03-07:
By selecting "Y", I can see that the forward gain is 20.31dB
  
[image]
  
Then if I rotate it to 77 degrees I can see that the first sidelobe is 10.26dB, or -10.05dB on the main lobe (which is why the published radiation pattern charts will often set 0dB as the reference maximum)

[image]

KJ7LNW, 2025-03-07:
I think I understand what you mean now, it sounds like you would like to see numerical annotations of the colors, is that correct?

g4ixt, 2025-03-08:
Yes that would work.  I was thinking of more like this top pic, but the lower pic might be easier for you to do and would also be a big help

[image]

KJ7LNW, 2025-03-09:
I pushed a new version give it a try. you will need to compile from the master branch.

g4ixt, 2025-03-09:
Thanks!  I will try it tomorrow.

Ian

g4ixt, 2025-03-10:
It's a good useful information addition.  However the numerical annotation of colours needs to be updated to match the changes in "gain scaling" so that the values are consistent.  At the moment the scale annotation is fixed but the radiation pattern colours change with "gain scaling".

KJ7LNW, 2025-03-10:
[image]

g4ixt, 2025-03-11:
excellent, and the 'rel' scale is a good addition.  I suggest making the 'rel' scale steps adapt to the graduation length like the 'abs' scale if you can, since in 'log' scaling it's quite compressed (see screenshot).

I believe that the NEC gain figures are isotropic, so for the 'abs' side (only) you should make it say 'dBi'

Thanks for maintaining xnec2c!

Ian

[image]

g4ixt, 2025-03-11:
...and in fact on the main screen box it should also say dBi

KJ7LNW, 2025-03-11:
> I suggest making the 'rel' scale steps adapt to the graduation length like the 'abs' scale if you can, since in 'log' scaling it's quite compressed (see screenshot).

Good idea, do you have any recommendations on the range and step size?  I hard coded some values that I thought would be useful, but your example has quite the range. 

Did you change  -60 to -40 in your build, or is -40 just the floor of this antenna?

> ...and in fact on the main screen box it should also say dBi

Good point

> Thanks for maintaining xnec2c!

You are welcome, Neoklis 5B4AZ really made a great program.

g4ixt, 2025-03-11:
-40 must be the floor, it's a pair of yagis spaced apart 3 wavelengths so there is probably a lower chance of very deep nulls
  
I suggest 6dB steps on the 'rel' axis for no better reason than 6db is one "S point" on a receiver, and steps down to -36dB.  I'd also suggest a '+' in front of the positive numbers on the dBi scale to separate them from the axis 'tick' that looks like a minus sign.

KJ7LNW, 2025-03-11:
Perhaps -6dB is a good option and I also want to include -3dB as the half-power beam-width, too

KJ7LNW, 2025-03-11:
These are applied, and I have also fixed a colors bug that has been around for a long time when selecting an arbitrary frequency (as opposed to one of the frequencies that was calculated by the FR card).

[image]

g4ixt, 2025-03-12:
Looking really good now, but I am unable to compile it.  ./configure gives this error:
  
checking for pthread_create in -lpthread... yes
./configure: line 5625: syntax error near unexpected token `clapack.h'
./configure: line 5625: `  _AC_CHECK_HEADERS_VSC(clapack.h cblas.h lapacke.h openblas/cblas.h openblas/lapacke.h, , , )'

I have looked at one of the previous configure scripts but it's a lot different in this area so I couldn't guess what the problem might be

KJ7LNW, 2025-03-12:
pull and try again. make sure you do ./autogen.sh

g4ixt, 2025-03-12:
Same error, plus others I didn't notice before.  existing xnec2c directory deleted first

(base) ian@Study:~$ git clone -b master https://github.com/KJ7LNW/xnec2c.git
Cloning into 'xnec2c'...
remote: Enumerating objects: 3276, done.
remote: Counting objects: 100% (620/620), done.
remote: Compressing objects: 100% (173/173), done.
remote: Total 3276 (delta 484), reused 479 (delta 446), pack-reused 2656 (from 2)
Receiving objects: 100% (3276/3276), 3.82 MiB | 25.05 MiB/s, done.
Resolving deltas: 100% (2502/2502), done.
(base) ian@Study:~$ cd xnec2c
(base) ian@Study:~/xnec2c$ ./autogen.sh
Preparing the xnec2c build system...please wait

Found GNU Autoconf version 2.71
Found GNU Automake version 1.16.5
Found GNU Libtool version 2.4.7

Automatically preparing build ... Warning: autoreconf failed
Attempting to run the preparation steps individually

Preparing build ... 
Warning:  Unsupported macros were found in ./configure.ac

The configure.ac file was scanned in order to determine if any
unsupported macros are used that exceed the minimum version
settings specified within this file.  As such, the following macros
should be removed from configure.ac or the version numbers in this
file should be increased:

AC_FUNC_REALLOC

Ignorantly continuing build preparation ... 
Warning: autoconf seems to have succeeded by removing the following options:
        AUTOCONF_OPTIONS="-f"

Removing those options should not be necessary and indicate some other
problem with the build system.  The build preparation is highly suspect
and may result in configuration or compilation errors.  Consider
rerunning the build preparation with verbose output enabled.
        ./autogen.sh --verbose

Continuing build preparation ... done

The xnec2c build system is now prepared.  To build here, run:
  ./configure
  make
(base) ian@Study:~/xnec2c$ ./configure
checking for a BSD-compatible install... /usr/bin/install -c
checking whether build environment is sane... yes
checking for a race-free mkdir -p... /usr/bin/mkdir -p
checking for gawk... no
checking for mawk... mawk
checking whether make sets $(MAKE)... yes
checking whether make supports nested variables... yes
checking whether to enable maintainer-specific portions of Makefiles... yes
checking build system type... x86_64-pc-linux-gnu
checking host system type... x86_64-pc-linux-gnu
checking whether configure should try to set CFLAGS... yes
checking for gcc... gcc
checking whether the C compiler works... yes
checking for C compiler default output file name... a.out
checking for suffix of executables... 
checking whether we are cross compiling... no
checking for suffix of object files... o
checking whether the compiler supports GNU C... yes
checking whether gcc accepts -g... yes
checking for gcc option to enable C11 features... none needed
checking whether gcc understands -c and -o together... yes
checking whether make supports the include directive... yes (GNU style)
checking dependency style of gcc... gcc3
checking whether make sets $(MAKE)... (cached) yes
checking for a sed that does not truncate output... /usr/bin/sed
checking for pkg-config... /usr/bin/pkg-config
checking pkg-config is at least version 0.9.0... yes
checking for glib-compile-resources tool... /usr/bin/glib-compile-resources
checking for GIO... yes
checking glib-compile-resources tool works... yes
checking glib-compile-resources tool output compiles and links... yes
checking for GMODULE... yes
checking for GTK... yes
checking for atan2 in -lm... yes
checking for GNU libc compatible malloc... yes
checking for GNU libc compatible realloc... yes
checking for pthread_create in -lpthread... yes
./configure: line 5631: syntax error near unexpected token `clapack.h'
./configure: line 5631: `  _ORIG_AC_CHECK_HEADERS(clapack.h cblas.h lapacke.h openblas/cblas.h openblas/lapacke.h, , , )'
(base) ian@Study:~/xnec2c$

g4ixt, 2025-03-12:
I don't know why it's striking through the lines above

g4ixt, 2025-03-12:
(base) ian@Study:~/xnec2c$ ./autogen.sh --verbose
Verbose output enabled
Found a configure template: ./configure.ac
Preparing the xnec2c build system...please wait

Checking autoreconf version: autoreconf --version
Checking autoconf version: autoconf --version
Found GNU Autoconf version 2.71
Checking if 2.71.0 is greater than 2.52.0
Checking automake version: automake --version
Found GNU Automake version 1.16.5
Checking if 1.16.5 is greater than 1.6.0
Checking libtoolize version: libtoolize --version
Found GNU Libtool version 2.4.7
Checking if 2.4.7 is greater than 1.4.2
Checking aclocal version: aclocal --version
Checking autoheader version: autoheader --version
Checking whether to only output version information
Backing up COPYING in /home/ian/xnec2c
cp -p COPYING COPYING.11479.protect_from_automake.backup
Backing up INSTALL in /home/ian/xnec2c
cp -p INSTALL INSTALL.11479.protect_from_automake.backup
Found an autom4te.cache directory, deleting it
rm -rf autom4te.cache
Detected auxiliary directory: auto-aux
mv -f "auto-aux/config.guess" "auto-aux/config.guess.backup"
mv -f "auto-aux/config.sub" "auto-aux/config.sub.backup"
Found extra aclocal search directory: m4

Automatically preparing build ... autoreconf  -I m4 -i -f
configure:5631: error: possibly undefined macro: _ORIG_AC_CHECK_HEADERS
      If this token and others are legitimate, please use m4_pattern_allow.
      See the Autoconf documentation.
autoreconf: error: /usr/bin/autoconf failed with exit status: 1
Warning: autoreconf failed
Attempting to run the preparation steps individually

Preparing build ... aclocal  -I m4 

Searching for AC_PROG_LIBTOOL in ./configure.ac
Searching for AM_PROG_LIBTOOL in ./configure.ac
Searching for LT_INIT in ./configure.ac

autoconf -f
configure:5631: error: possibly undefined macro: _ORIG_AC_CHECK_HEADERS
      If this token and others are legitimate, please use m4_pattern_allow.
      See the Autoconf documentation.
Searching for AC_C_RESTRICT in ./configure.ac
Searching for AC_INCLUDES_DEFAULT in ./configure.ac
Searching for AC_LANG_ASSERT in ./configure.ac
Searching for AC_LANG_WERROR in ./configure.ac
Searching for AS_SET_CATFILE in ./configure.ac
Searching for AC_COMPILER_IFELSE in ./configure.ac
Searching for AC_FUNC_MBRTOWC in ./configure.ac
Searching for AC_HEADER_STDBOOL in ./configure.ac
Searching for AC_LANG_CONFTEST in ./configure.ac
Searching for AC_LANG_SOURCE in ./configure.ac
Searching for AC_LANG_PROGRAM in ./configure.ac
Searching for AC_LANG_CALL in ./configure.ac
Searching for AC_LANG_FUNC_TRY_LINK in ./configure.ac
Searching for AC_MSG_FAILURE in ./configure.ac
Searching for AC_PREPROC_IFELSE in ./configure.ac
Searching for AC_C_BACKSLASH_A in ./configure.ac
Searching for AC_CONFIG_LIBOBJ_DIR in ./configure.ac
Searching for AC_GNU_SOURCE in ./configure.ac
Searching for AC_PROG_EGREP in ./configure.ac
Searching for AC_PROG_FGREP in ./configure.ac
Searching for AC_REPLACE_FNMATCH in ./configure.ac
Searching for AC_FUNC_FNMATCH_GNU in ./configure.ac
Searching for AC_FUNC_REALLOC in ./configure.ac
Searching for AC_TYPE_MBSTATE_T in ./configure.ac

Warning:  Unsupported macros were found in ./configure.ac

The configure.ac file was scanned in order to determine if any
unsupported macros are used that exceed the minimum version
settings specified within this file.  As such, the following macros
should be removed from configure.ac or the version numbers in this
file should be increased:

AC_FUNC_REALLOC

Ignorantly continuing build preparation ... 
autoconf

Warning: autoconf seems to have succeeded by removing the following options:
        AUTOCONF_OPTIONS="-f"

Removing those options should not be necessary and indicate some other
problem with the build system.  The build preparation is highly suspect
and may result in configuration or compilation errors.  Consider
reviewing the minimum GNU Autotools version settings contained in
this script along with the macros being used in your configure.ac file.

Continuing build preparation ... Searching for AM_CONFIG_HEADER in ./configure.ac
Searching for AC_CONFIG_HEADER in ./configure.ac
autoheader 

Searching for AM_INIT_AUTOMAKE in ./configure.ac
automake -a -c -f
configure.ac:66: installing 'auto-aux/config.guess'
configure.ac:66: installing 'auto-aux/config.sub'
rm -f COPYING.11479.protect_from_automake.backup

Restoring INSTALL from backup (automake -f likely clobbered it)
rm -f INSTALL
mv INSTALL.11479.protect_from_automake.backup INSTALL
rm -f INSTALL.11479.protect_from_automake.backup
rm -f "auto-aux/config.guess.backup"
rm -f "auto-aux/config.sub.backup"
done

The xnec2c build system is now prepared.  To build here, run:
  ./configure
  make
rm -f "auto-aux/config.guess.backup"
rm -f "auto-aux/config.sub.backup"
(base) ian@Study:~/xnec2c$

KJ7LNW, 2025-03-13:
Sorry for the hassle, I reverted that build tweak, so it should be back to the way it was. 

I was hoping that configure would auto-build vscode's header include references, but it seems more complicated than I would like.

g4ixt, 2025-03-13:
[image]

Excellent, working fine now, thank you again!

KJ7LNW, 2025-03-13:
You are welcome!  

I noticed that your minimum is -999, we could be about because I thought I had that clamped to -60. do you change the code there?  If not, may I get a copy of your .NEC to investigate ?

g4ixt, 2025-03-14:
No, I haven't changed the code - although I write Python code I have no idea about C.  In the file attached, the RP card is set to just show the radiation pattern if you click 'Z'.  If you rotate it to zero degrees the gain figure jumps to -999 but only then.

[23ele_v2.tar.gz](https://github.com/user-attachments/files/19244844/23ele_v2.tar.gz)

KJ7LNW, 2025-12-19:
I think this is done, so I'm going to close the ticket. I just released version 4.4.18 if you would like to give it a try.

closed 2025-12-19 by KJ7LNW