KJ7LNW/xnec2c #64: install for ubuntu 24.04 doesn't complete

jeffbob970, 2025-04-25:
the make install results in 

Making install in src
make[1]: Entering directory '/home/dr_jeff/ham_stuff/xnec2c_inputs/xnec2c/src'
make[2]: Entering directory '/home/dr_jeff/ham_stuff/xnec2c_inputs/xnec2c/src'
 /usr/bin/mkdir -p '/usr/local/bin'
  /usr/bin/install -c xnec2c '/usr/local/bin'
/usr/bin/install: cannot create regular file '/usr/local/bin/xnec2c': Permission denied
make[2]: *** [Makefile:469: install-binPROGRAMS] Error 1
make[2]: Leaving directory '/home/dr_jeff/ham_stuff/xnec2c_inputs/xnec2c/src'
make[1]: *** [Makefile:1106: install-am] Error 2
make[1]: Leaving directory '/home/dr_jeff/ham_stuff/xnec2c_inputs/xnec2c/src'
make: *** [Makefile:803: install-recursive] Error 1

Any help will be appreciated

KJ7LNW, 2025-04-25:
you probably need to be root, try `sudo make install`

alternatively you can run it directly from the build directory if you do not wish to install it in the operating system by running `./src/xnec2c -j $(nproc)`

jeffbob970, 2025-04-25:
That works like a champ.  I had sudo make && make install. It did want the sudo  before the make install too. I also had to put the input file in the src directory. I imagine if I weren't lazy and put the who path to the input file I wouldn't have had to do that.

Excellent software! Excellent response time and response.

Thank you

KJ7LNW, 2025-04-25:
I am glad it worked. Neoklis did a great job writing xnec2c, for sure!

KJ7LNW, 2025-04-25:
fyi, if you had exactly `sudo make && make install` then it will do this, breaking at `&&`:

```sh
sudo make
make install
```

so you may have wanted `sudo make && sudo make install` if you wanted to do the full installation

closed 2025-04-25 by KJ7LNW