KJ7LNW/xnec2c #65: [err] failed to add objects to builder: The resource at “/xnec2c.glade” does not exist

fred-corp, 2025-06-04:
I'm trying to build xnec2c on macOS. It seems to build fine, but when I try to launch xnec2c, I get the following errors : 

```
$ ./src/xnec2c -j4 examples/airplane.nec 
[err] failed to add objects to builder: The resource at “/xnec2c.glade” does not exist
[err] Listing Object Ids:
[err] main_menu
[err] main_freq_adjustment
[err] main_incline_adjustment
[err] main_rotate_adjustment
[err] main_zoom_adjustment
[err] main_window
[err] menuitem1_menu
[err] main_new
[err] main_open_input
[err] nec2_edit
[err] main_save
[err] main_save_as
[err] struct_save_as_gnuplot
[err] optimizer_output
[err] confirm_quit
[err] main_quit
[err] main_view_menuitem
[err] main_view_menuitem_menu
[err] main_rdpattern
[err] main_freqplots
[err] main_pol_menu
[err] main_pol_menu_menu
[err] main_total
[err] main_horizontal
[err] main_vertical
[err] main_right_hand
[err] main_left_hand
[err] main_common_projection
[err] main_common_frequency
[err] menuitem4_menu
[err] about
[err] main_hbox
[err] main_hbox1
[err] main_currents_togglebutton
[err] main_charges_togglebutton
[err] main_loop_start
[err] main_loop_pause
[err] main_loop_reset
[err] structure_fstep_entry
[err] main_hbox2
[err] main_colorcode_zerolabel
[err] main_colorcode_drawingarea
[err] main_colorcode_maxlabel
[err] main_gain_entry
[err] main_freq_spinbutton
[err] main_freq_label
[err] main_freq_checkbutton
[err] main_new_freq_button
[err] main_grid1
[err] main_x_axis
[err] main_y_axis
[err] main_z-axis
[err] main_default_view
[err] main_rotate_spinbutton
[err] main_incline_spinbutton
[err] saveas_gnuplot
[err] main_zoom_spinbutton
[err] structure_plus_button
[err] structure_minus_button
[err] structure_one_button
[err] structure_drawingarea
[err] structure_frame
[err] struct_label
read(): Invalid argument
[err] child 3  length 7  return 0
read(): Invalid argument                                                                                                                                         
[err] child 2  length 7  return 0
read(): Invalid argument
[err] child 1  length 7  return 0
read(): Invalid argument
[err] child 0  length 7  return 0
```

I've attached the `config.log` file.  
Can anyone help me to solve this issue ?

[config.log](https://github.com/user-attachments/files/20589517/config.log)

KJ7LNW, 2025-06-05:
It looks like you must have built a very old version of xnec2c because the current version builds `xnec2c.glade` into the binary.

Did you already try the following  from https://www.xnec2c.org/ ?

- git clone https://github.com/KJ7LNW/xnec2c.git
- cd xnec2c
- ./autogen.sh
- ./configure
- make && make install
- make desktop-install # Optional: for icons and file association.
- xnec2c

KJ7LNW, 2025-12-19:
I'm assuming this was a build problem. Please make sure you're running the latest build. I just released version 4.4.18. So go ahead and give it a try!