KJ7LNW/xnec2c #68: Problem with the GM cards

likecabezon, 2025-09-26:
When using the GM card to transform a single geometry and not duplicate you use in the "New Structures" menu a 0 to indicate you only want to transform the selected structures. But when saving the file the program complains about "Segment tag number error"

NEC2 documentation: https://www.nec2.org/part_3/cards/gm.html
Even if your program is not supposed to support the nec2 syntax you still need a way of transforming geometries. For example now is not possible to generate a loop antenna parallel to the GND plane without entering manually all the segments. 
Thanks for bringing this GUI for the nec2 world :).

KJ7LNW, 2025-09-26:
Can you provide me with the sample file that is having trouble?

KJ7LNW, 2025-09-30:
The information you provided has allowed me to make the error messages more informative. Hopefully, this will help somebody in the future, so thank you for your contribution!

```diff
-    Stop( _("Segment tag number error.  You are (probably) referencing an unknown tag or\n"
-            "segment number in a control card."), ERR_OK );
   
+    if( itagi == 0 )
+    {
+      Stop( _("Segment tag number error: Tag 0 is reserved and cannot be used.\n"
+              "Check your geometry cards (especially GM cards) to ensure tag values are valid.\n"
+              "For GM cards: verify 'start from tag' and 'tag increment' don't create tag 0."), 
+ ERR_OK );
+    }
+    else
+    {
+      Stop( _("Segment tag number error: No segment found with tag %d.\n"
+              "You are referencing an unknown tag or segment in a control card.\n"
+              "Valid tags in your geometry are those assigned to GW/GA/GH/SP cards."), ERR_OK, 
+ itagi );
+    }
+  
     return( -1 );
   }
```

closed 2025-09-30 by KJ7LNW